dariuszseweryn/RxAndroidBle #749: Android 12 Developer Preview 3

ArthurMytsko, 2021-05-07:
**Describe the bug**
Application crash when try scan devices when ScanFilter has `null` device mac address

**To Reproduce**
Steps to reproduce the behavior:
1. Create a `ScanFilter` and don't specify device mac address
2. Perform `RxBleClient.scanBleDevices(...)`
3. `NullPointerException` occurs

**Expected behavior**
Application start scanning bluetooth devices

**Smartphone (please complete the following information):**
 - Device: Google Pixel 4XL
 - OS: Android 12 Developer Preview 3
 - Library version: 1.12.1
 
**Exception stacktrace**

```
java.lang.NullPointerException
    at com.android.internal.util.Preconditions.checkNotNull(Preconditions.java:147)
    at android.bluetooth.le.ScanFilter$Builder.setDeviceAddressInternal(ScanFilter.java:678)
    at android.bluetooth.le.ScanFilter$Builder.setDeviceAddress(ScanFilter.java:615)
    at android.bluetooth.le.ScanFilter$Builder.setDeviceAddress(ScanFilter.java:590)
    at com.polidea.rxandroidble2.internal.scan.AndroidScanObjectsConverter.toNative(AndroidScanObjectsConverter.java:71)
    at com.polidea.rxandroidble2.internal.scan.AndroidScanObjectsConverter.toNativeFilters(AndroidScanObjectsConverter.java:56)
    at com.polidea.rxandroidble2.internal.operations.ScanOperationApi21.startScan(ScanOperationApi21.java:117)
    at com.polidea.rxandroidble2.internal.operations.ScanOperationApi21.startScan(ScanOperationApi21.java:28)
    at com.polidea.rxandroidble2.internal.operations.ScanOperation.protectedRun(ScanOperation.java:43)
    at com.polidea.rxandroidble2.internal.QueueOperation$1.subscribe(QueueOperation.java:41)
    at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
    at io.reactivex.Observable.subscribe(Observable.java:12284)
    at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn.subscribeActual(ObservableUnsubscribeOn.java:32)
    at io.reactivex.Observable.subscribe(Observable.java:12284)
    at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1.run(FIFORunnableEntry.java:56)
    at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
    at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
    at java.util.concurrent.FutureTask.run(FutureTask.java:266)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
    at java.lang.Thread.run(Thread.java:920)
```

**Additional context**
The issue is related to OS only and should not be realted to device of library version (at least few last ones)
The problem here is inside `AndroidScanObjectsConverter.toNative(scanFilter)`
As the `Device Address` inside `ScanFilter` is `null`, the sdk's `ScanFilter.Builder` verifies that the value is not null and throws exception. 
`toNative()` method should set only `not null` values to the builder. Seems that `null` chedking was added in Android 12 DP 3

ArthurMytsko, 2021-05-07:
BTW, Android 12 DP3 now doesn't requre to have LOCATION permission. they have added new ones like [Manifest.permission.BLUETOOTH_CONNECT](https://developer.android.com/reference/android/Manifest.permission#BLUETOOTH_CONNECT) and [Manifest.permission.BLUETOOTH_SCAN](https://developer.android.com/reference/android/Manifest.permission#BLUETOOTH_SCAN)
They haven't shared release notes about it, but the chagnes are present in SDK and also new permissions are requred during the application work (already tested it)

ArthurMytsko, 2021-05-07:
It will be good to start adopting this librarly for Android 12 needs.

ArthurMytsko, 2021-05-11:
@dariuszseweryn  Sure, I can help with testing. I'm working on a project which depends on your library, so I have an interest to make it work with Android 12 :)

dariuszseweryn, 2021-05-13:
@ArthurMytsko You may try Jitpack against branch `fix/android_12_scan_filter_null_mac_address` and check if it fixes the issue and report back :)

dariuszseweryn, 2021-05-14:
As I see this should not be a problem in the latest version of [the code](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/le/ScanFilter.java#589)?
```
/**
  | * Set filter on device address.
  | *
  | * @param deviceAddress The device Bluetooth address for the filter. It needs to be in the
  | * format of "01:02:03:AB:CD:EF". The device address can be validated using {@link
  | * BluetoothAdapter#checkBluetoothAddress}.  The @AddressType is defaulted to {@link
  | * BluetoothDevice#ADDRESS_TYPE_PUBLIC}
  | * @throws IllegalArgumentException If the {@code deviceAddress} is invalid.
  | */
  | public Builder setDeviceAddress(String deviceAddress) {
  |     if (deviceAddress == null) {
  |         mDeviceAddress = deviceAddress;
  |         return this;
  |     }
  |     return setDeviceAddress(deviceAddress, BluetoothDevice.ADDRESS_TYPE_PUBLIC);
  | }
```

ArthurMytsko, 2021-05-14:
> As I see this should not be a problem in the latest version of [the code](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/le/ScanFilter.java#589)?
>
> ``` [...]

Android 12 source code is not published yet. most probably the link you sent is Andorid 11

ArthurMytsko, 2021-05-14:
> @ArthurMytsko You may try Jitpack against branch `fix/android_12_scan_filter_null_mac_address` and check if it fixes the issue and report back :)

That fix should work, previously I've tried it locally on sample app

dariuszseweryn, 2021-05-14:
The code is not published to be available via Android Studio but otherwise it is public. The [S-preview-1 version of the code](https://android.googlesource.com/platform/frameworks/base/+/refs/tags/android-s-preview-1/core/java/android/bluetooth/le/ScanFilter.java#552) (which is the only one tagged with "s") looks different. Plus Android 11 didn't have a `ADDRESS_TYPE_PUBLIC` (shame that is not usable from the user's perspective yet)

Have you seen any official information about `Manifest.permission.BLUETOOTH_CONNECT` and `Manifest.permission.BLUETOOTH_SCAN`? How do they behave? Are Location Services still needed if the app wants to scan?

ArthurMytsko, 2021-05-14:
The change regarding Device address was added in Developer Preview 3 (at least the app was working fine with DP2)

Regarding new permissions - for some reason google haven't anonced it officially. I've found it in API changes. The only information about them could be found in API reference (links I posted previously)

No, Location service is not needed to scan Bluetooth devices. Manifest.permission.BLUETOOTH_SCAN is enough to scan and connect to the device.

remonh87, 2021-05-15:
I am using the library as a base for my Flutter library. I can also help out testing the lib and logging/fixing issues.

j-roskopf, 2021-05-20:
👋 hello -

I pulled down the `fix/android_12_scan_filter_null_mac_address` branch to try it in our app and I am receiving the following exception on my Android 12 device. Please let me know if there is any additional info I can help provide!

```    java.lang.NoClassDefFoundError: Failed resolution of: Lcom/jakewharton/rxrelay2/BehaviorRelay;
        at com.polidea.rxandroidble2.internal.DeviceModule.provideConnectionStateRelay(DeviceModule.java:57)
        at com.polidea.rxandroidble2.internal.DeviceModule_ProvideConnectionStateRelayFactory.get(DeviceModule_ProvideConnectionStateRelayFactory.java:17)
        at com.polidea.rxandroidble2.internal.DeviceModule_ProvideConnectionStateRelayFactory.get(DeviceModule_ProvideConnectionStateRelayFactory.java:9)
        at bleshadow.dagger.internal.DoubleCheck.get(DoubleCheck.java:47)
        at com.polidea.rxandroidble2.internal.RxBleDeviceImpl_Factory.get(RxBleDeviceImpl_Factory.java:31)
        at com.polidea.rxandroidble2.internal.RxBleDeviceImpl_Factory.get(RxBleDeviceImpl_Factory.java:11)
        at bleshadow.dagger.internal.DoubleCheck.get(DoubleCheck.java:47)
        at com.polidea.rxandroidble2.DaggerClientComponent$DeviceComponentImpl.provideDevice(DaggerClientComponent.java:509)
        at com.polidea.rxandroidble2.internal.RxBleDeviceProvider.getBleDevice(RxBleDeviceProvider.java:43)
        at com.polidea.rxandroidble2.internal.scan.InternalToExternalScanResultConverter.apply(InternalToExternalScanResultConverter.java:26)
        at com.polidea.rxandroidble2.internal.scan.InternalToExternalScanResultConverter.apply(InternalToExternalScanResultConverter.java:13)
        at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:57)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onNext(ObservableUnsubscribeOn.java:60)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.onNext(ObservableCreate.java:66)
        at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1$1.onNext(FIFORunnableEntry.java:68)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onNext(ObservableUnsubscribeOn.java:60)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.onNext(ObservableCreate.java:66)
        at com.polidea.rxandroidble2.internal.operations.ScanOperationApi21$1.onScanResult(ScanOperationApi21.java:83)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:495)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loopOnce(Looper.java:201)
        at android.os.Looper.loop(Looper.java:288)
        at android.app.ActivityThread.main(ActivityThread.java:7727)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "com.jakewharton.rxrelay2.BehaviorRelay" on path: DexPathList[[zip file "/data/app/~~L_Ht2APB6tyHIvWNSunmPQ==/PACKAGENAME-zgPpgInIuWiwKZ8rlNeMng==/base.apk"],nativeLibraryDirectories=[/data/app/~~L_Ht2APB6tyHIvWNSunmPQ==/PACKAGENAME-zgPpgInIuWiwKZ8rlNeMng==/lib/arm64, /data/app/~~L_Ht2APB6tyHIvWNSunmPQ==/PACKAGENAME-zgPpgInIuWiwKZ8rlNeMng==/base.apk!/lib/arm64-v8a, /system/lib64, /system_ext/lib64, /product/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:207)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at com.polidea.rxandroidble2.internal.DeviceModule.provideConnectionStateRelay(DeviceModule.java:57) 
        at com.polidea.rxandroidble2.internal.DeviceModule_ProvideConnectionStateRelayFactory.get(DeviceModule_ProvideConnectionStateRelayFactory.java:17) 
        at com.polidea.rxandroidble2.internal.DeviceModule_ProvideConnectionStateRelayFactory.get(DeviceModule_ProvideConnectionStateRelayFactory.java:9) 
        at bleshadow.dagger.internal.DoubleCheck.get(DoubleCheck.java:47) 
        at com.polidea.rxandroidble2.internal.RxBleDeviceImpl_Factory.get(RxBleDeviceImpl_Factory.java:31) 
        at com.polidea.rxandroidble2.internal.RxBleDeviceImpl_Factory.get(RxBleDeviceImpl_Factory.java:11) 
        at bleshadow.dagger.internal.DoubleCheck.get(DoubleCheck.java:47) 
        at com.polidea.rxandroidble2.DaggerClientComponent$DeviceComponentImpl.provideDevice(DaggerClientComponent.java:509) 
        at com.polidea.rxandroidble2.internal.RxBleDeviceProvider.getBleDevice(RxBleDeviceProvider.java:43) 
        at com.polidea.rxandroidble2.internal.scan.InternalToExternalScanResultConverter.apply(InternalToExternalScanResultConverter.java:26) 
        at com.polidea.rxandroidble2.internal.scan.InternalToExternalScanResultConverter.apply(InternalToExternalScanResultConverter.java:13) 
        at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:57) 
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onNext(ObservableUnsubscribeOn.java:60) 
[... cut ...]
        at android.os.Handler.handleCallback(Handler.java:938) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loopOnce(Looper.java:201) 
        at android.os.Looper.loop(Looper.java:288) 
        at android.app.ActivityThread.main(ActivityThread.java:7727) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975) 
```

ArthurMytsko, 2021-05-20:
Android 12 Beta 1 has official docs about [bluetooth permissions](https://developer.android.com/about/versions/12/features/bluetooth-permissions)

j-roskopf, 2021-05-25:
@dariuszseweryn were there any thoughts on when this fix might be released?

dariuszseweryn, 2021-05-25:
[This shouldn't be a problem on Android S Beta 1 anymore](https://android.googlesource.com/platform/frameworks/base/+/refs/tags/android-s-beta-1/core/java/android/bluetooth/le/ScanFilter.java#590) so I guess the fix will be obsolete. Are you able to test on the beta?

j-roskopf, 2021-05-25:
I am currently testing on the beta and am still seeing that same exception in the original post

```    com.polidea.rxandroidble2.exceptions.BleScanException: Bluetooth cannot start (code 0)
        at com.polidea.rxandroidble2.internal.operations.ScanOperation.protectedRun(ScanOperation.java:50)
        at com.polidea.rxandroidble2.internal.QueueOperation$1.subscribe(QueueOperation.java:41)
        at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
        at io.reactivex.Observable.subscribe(Observable.java:12284)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn.subscribeActual(ObservableUnsubscribeOn.java:32)
        at io.reactivex.Observable.subscribe(Observable.java:12284)
        at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1.run(FIFORunnableEntry.java:56)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:920)
     Caused by: java.lang.NullPointerException
        at com.android.internal.util.Preconditions.checkNotNull(Preconditions.java:147)
        at android.bluetooth.le.ScanFilter$Builder.setDeviceAddressInternal(ScanFilter.java:678)
        at android.bluetooth.le.ScanFilter$Builder.setDeviceAddress(ScanFilter.java:615)
        at android.bluetooth.le.ScanFilter$Builder.setDeviceAddress(ScanFilter.java:590)
        at com.polidea.rxandroidble2.internal.scan.AndroidScanObjectsConverter.toNative(AndroidScanObjectsConverter.java:71)
        at com.polidea.rxandroidble2.internal.scan.AndroidScanObjectsConverter.toNativeFilters(AndroidScanObjectsConverter.java:56)
        at com.polidea.rxandroidble2.internal.operations.ScanOperationApi21.startScan(ScanOperationApi21.java:104)
        at com.polidea.rxandroidble2.internal.operations.ScanOperationApi21.startScan(ScanOperationApi21.java:28)
        at com.polidea.rxandroidble2.internal.operations.ScanOperation.protectedRun(ScanOperation.java:43)
        at com.polidea.rxandroidble2.internal.QueueOperation$1.subscribe(QueueOperation.java:41) 
        at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40) 
        at io.reactivex.Observable.subscribe(Observable.java:12284) 
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn.subscribeActual(ObservableUnsubscribeOn.java:32) 
        at io.reactivex.Observable.subscribe(Observable.java:12284) 
        at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1.run(FIFORunnableEntry.java:56) 
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38) 
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26) 
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:920) 
```

ArthurMytsko, 2021-05-27:
> [This shouldn't be a problem on Android S Beta 1 anymore](https://android.googlesource.com/platform/frameworks/base/+/refs/tags/android-s-beta-1/core/java/android/bluetooth/le/ScanFilter.java#590) so I guess the fix will be obsolete. Are you able to test on the beta?

With Android 12 Beta 1 app fails with the same stacktrace

dariuszseweryn, 2021-05-27:
There is a different issue in regards to the new Bluetooth Permissions #752

closed 2021-05-27 by dariuszseweryn

dariuszseweryn, 2021-05-27:
Hopefully a 1.12.2 version snapshot (see readme) will be available shortly

PR #754 (2021-05-27): Added conditional MAC address setting
Apparently API 31 (S) does check for null MAC address when specifying a ScanFilter.
This fix will likely be obsolete on some next version of the Android 12 when it will be available for tests.
Closes #749