dariuszseweryn/RxAndroidBle #751: Deprecated features in gradle files

lcrocker, 2021-05-20:
Can't build Android library with recent versions of gradle.

Just cloned the repo, and tried to build any target with ./gradlew (in particular, I want to build the "jar" target so I can make some changes, and use my version in Multiplatform, and then use that in FlutterBleLib...)

I get "maven plugin is deprecated...use the maven-publish plugin instead...kotlin-android-extensions is deprecated...use migration guide..."

I tried gradle 6.7.1 and 7.0.2.  Here's the output of gradle --version:

------------------------------------------------------------
Gradle 7.0.2
------------------------------------------------------------

Build time:   2021-05-14 12:02:31 UTC
Revision:     1ef1b260d39daacbf9357f9d8594a8a743e2152e

Kotlin:       1.4.31
Groovy:       3.0.7
Ant:          Apache Ant(TM) version 1.10.9 compiled on September 27 2020
JVM:          11.0.11 (Ubuntu 11.0.11+9-Ubuntu-0ubuntu2.20.04)
OS:           Linux 5.8.0-53-generic amd64

Alternatively, is there a way to build a jar file with Android Studio to then propagate up the chain? Or is there a way to maybe use the dex files produced by the AS build in Multi and Flutter?

My ultimate goal is a Flutter app...but I need to add a bit of my own Java code to Rx. I'm happy to do it, but if I can't compile, I'm out of luck...

dariuszseweryn, 2021-05-25:
Hi, what is the command you execute? Deprecation does not mean it is already not supported and you have not provided an error you encounter. AFAIK you cannot package an Android library to a jar, only aar is supported.

lcrocker, 2021-05-25:
Thanks for the response.

I'm trying to do what should be a very simple thing in most other languages and build systems: clone the Git repos exactly as they are, make a small change to one source file, then rebuild the repos. I have forked RxAndroidBle, MultiPlatformBleAdapter, and FlutterBleLib but made no changes yet. I downloaded the up-to-the-minute version of Android Studio, and I'm running Gradle 6.7.1, and Maven 3.6.3. All of this on Ubuntu 20.04 LTS, with OpenJDK 11.

So, I first cloned RxAndroidBle, master branch. Running "./gradlew build" produces screenfuls of errors:

/home/lee/work/RxAndroidBle/mockrxandroidble/src/main/java/com/polidea/rxandroidble2/mockrxandroidble   /RxBleConnectionMock.java:5: error: cannot find symbol
import androidx.annotation.NonNull;
                          ^
  symbol:   class NonNull
  location: package androidx.annotation
/home/lee/work/RxAndroidBle/mockrxandroidble/src/main/java/com/polidea/rxandroidble2/mockrxandroidble/RxBleConnectionMock.java:10: error: cannot find symbol
import com.polidea.rxandroidble2.ConnectionParameters;
                                ^
  symbol:   class ConnectionParameters
  location: package com.polidea.rxandroidble2
/home/lee/work/RxAndroidBle/mockrxandroidble/src/main/java/com/polidea/rxandroidble2/mockrxandroidble/RxBleConnectionMock.java:11: error: cannot find symbol
import com.polidea.rxandroidble2.NotificationSetupMode;
                                ^
. . .

  symbol:   class NotificationSetupMode
  location: class RxBleConnectionMock
/home/lee/work/RxAndroidBle/mockrxandroidble/src/main/java/com/polidea/rxandroidble2/mockrxandroidble/RxBleConnectionMock.java:588: error: cannot find symbol
    private Observable<Observable<byte[]>> createCharacteristicNotificationObservable(final UUID characteristicUuid,
            ^
  symbol:   class Observable
  location: class RxBleConnectionMock
100 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':mockrxandroidble:androidJavadocs'.
> Javadoc generation failed. Generated Javadoc options file (useful for troubleshooting): '/home/lee/work/RxAndroidBle/mockrxandroidble/build/tmp/androidJavadocs/javadoc.options'

========

So, I tried Android Studio. It loaded the project, I pressed the little hammer icon, and it said "Gradle build finished in 10s". But then "find . -name '*.aar'" does not find any .aar files, so I'm not sure what it built or where.

So, moving on to MultiPlatformBleAdapter, it too fails attempts at gradle but builds something with Studio. So I tried to do what I'll need for phase 2: I'll need Multi to use an RxAndroidBle .aar built from my fork rather than the one at JCenter. But since I couldn't build one yet, I just downloaded the latest from JCenter and put it into a libs directory, added a "flatDIr" entry to repositories, and changed the implementation line in dependencies to "implementation files('libs/rxandroidble-1.12.1.aar'). That gave me errors that it couldn't find some of the things that are marked as dependencies of Rx in Maven, so I added jars for rxjava and rxrelay to the libs dir as well. Now it complains that it can't find something called "Package", but I can't find that text in any file of the repo, so I'm stuck there. Once again I abandoned Gradle and tried to do this through the Studio GUI ("Module Settings") but no luck.

I am working at a company where I have access to other kinds of computers. If you can tell me what kind of setup actually built these products for Android, I can go get an Apple machine or a Windows machine if I have to. My personal experience with other languages and build systems, though, is that Linux is usually the simplest and most reliable.

Anyway, I'd be grateful any build help you can give me, even if I have to resort to Studio. If we can get it running with gradle, I might even help you create some CI scripts in Github to run builds.

dariuszseweryn, 2021-05-27:
Android Studio comes with a packaged JDK 8 AFAIK and that is what Android officially supports – you may want to start with that.
Usually the library is installed by Gradle and it takes care of fetching other dependencies which are built along (shadowed Dagger 2, RxRelay 2, RxJava 2). I am having some difficulties in running the project myself using the latest Android Studio tbh — now it is only my pet project and I do not have enough time to handle everything so any help is appreciated.

lcrocker, 2021-05-27:
Thanks. I'll try a clean machine with JDK8.

dariuszseweryn, 2021-05-27:
TIL. New Android Studio does not list Gradle Tasks by default (need to change the setting under experimental) and it seems that Robolectric is now incompatible when run under AS. I foresee lots of maintenance in the near future

nrbrook, 2021-05-27:
@dariuszseweryn I looked at upgrading, but electricspock is the limiting factor. I tried to update its dependencies but it is very complex and I just got lost https://github.com/hkhc/electricspock/pull/11#issuecomment-774475212
After that comment I determined that JUnit 5 was not possible (I think due to JDK requirement or something). In the end it may be easier to remove Spock from the unit tests

dariuszseweryn, 2021-05-28:
Moving the entire test suite from Spock to something else is a major investment I think. I am looking into making stubs of used Android dependency classes in tests and getting rid of ElectricSpock and Robolectric. This could be less demanding

nrbrook, 2021-05-28:
Possibly, you would need to do the same for RxAndroid

dariuszseweryn, 2021-05-28:
RxAndroid does not seem to be used in the tests?

nrbrook, 2021-05-28:
Ah I thought the library used RxAndroid, but I guess not.

dariuszseweryn, 2021-05-28:
RxAndroid is used only in example apps. They are not tested at all, so no problem there ;D
We would only need to stub used classes of Android OS in parts of the code that are tested. I guess these should be only Context and Bluetooth related <10 classes. Hard to say how much burden it will be in the future but probably less than relying on an exotic testing framework

lcrocker, 2021-06-24:
On a clean machine with Oracle JDK8, Gradle 6.5, and the Android tools, I am able to build the AARs for this, as well as for MultiPlatform. "gradlew build" still fails because it tries to run the "lint" task.

So now I can build the AARs, I can build this one with the minimal change I need to support connectGatt(...phy). Now the issue will be getting the FlutterBleLib code to use my modified lib.

Darius, if this is something you could do relatively easily, my boss would be happy to throw some money and a phone your way.

lcrocker, 2021-09-17:
FYI, there's a StackOverflow bounty on my workaround question:

https://stackoverflow.com/questions/69184440/replacing-a-gradle-dependency-in-flutter-library

james1050, 2021-11-17:
I've managed to build RxAndroidBle with custom code and include it in a React Native project starting with a basis of react-native-ble-plx (v2.0.3) -> MultiPlatformBleAdapter (commit b0272c0fae040cc90f790481c54e8f2d45304339) -> RxAndroidBle (v1.7.1 RxJava1, commit a24ca50eecccac72ccb7bc6534641f436571e59f).

At first I tried to build everything at once in a multiproject including this old RxAndroidBle dependency. No success. Therefore I changed to building the RxAndroidBle separately using older Java from the command line, and added the .aar file to the MultiPlatformBleAdapter build.gradle.

The following actions are what worked for my React Native project. Hopefully they can be adapted for your Flutter project.

1. Submodule react-native-ble-plx at v2.0.3 (commit 70a254f32e24e017d1d2d0ee3a7c3c902de61f19) into your project
2. Submodule MultiPlatformBleAdapter (commit b0272c0fae040cc90f790481c54e8f2d45304339) into your project
3. Submodule RxAndroidBle (commit a24ca50eecccac72ccb7bc6534641f436571e59f) into your project
4. Add MultiPlatformBleAdapter to your app's settings.gradle similar to:
`include ':MultiPlatformBleAdapter'`
`project(':MultiPlatformBleAdapter').projectDir = new File(rootProject.projectDir, '../lib/ble/MultiPlatformBleAdapter/android/library')`
5. Change the following in react-native-ble-plx's android/build.gradle:
~~implementation 'com.github.Polidea:MultiPlatformBleAdapter:b0272c0fae040cc90f790481c54e8f2d45304339'~~
implementation project(':MultiPlatformBleAdapter')
6. I also changed to androidx instead of the support library for NonNull and Nullable in MultiPlatformBleAdapter but I'm no expert on Android projects to know this was actually necessary.

At this point your project should build with the local source version of MultiPlatformBleAdapter, but still downloading the Maven Polidea.RxAndroidBle. The following steps will finally use a locally build .aar file:

7. Install Java SE 8.
8. Modify the gradlew script in RxAndroidBle to set JAVA_HOME=<place you installed Java 8 jre> e.g. C:\Program Files\Java\jre1.8.0_202
9. I run into a build issue with the RxAndroidBle.sample project. Changing the minSdkVersion from 18 to 21 in the sample/build.gradle file fixes this.
10. Run the RxAndroidBle gradlew script with the ad hoc JAVA_HOME. This should now successfully build the whole RxAndroidBle project, including the needed .aar file at RxAndroidBle/rxandroidble/build/outputs/aar/rxandroidble-release.aar
11. Make the following changes to your MultiPlatformBleAdapter/android/library/build.gradle, within the dependencies block:
~~implementation "com.polidea.rxandroidble:rxandroidble:1.7.1"~~
implementation "com.android.support:support-annotations:27.1.0"
implementation "com.artemzin.rxjava:proguard-rules:1.3.3.0"
implementation "com.jakewharton.rxrelay:rxrelay:1.2.0"
implementation "com.polidea.rxandroidble:dagger-library-shadow:1.7.1"
implementation "io.reactivex:rxjava:1.3.8"
implementation files('<path to your .aar file/rxandroidble-release.aar>')

I think that was everything needed to get a build working.

nrbrook, 2021-11-17:
This issue is resolved with https://github.com/dariuszseweryn/RxAndroidBle/pull/763

dariuszseweryn, 2022-05-28:
We are now building with Gradle 7. Unfortunately there will be an issue when Gradle 8 comes but doesn't seem to be a big problem:
> Configure project :mockrxandroidble
The AbstractCompile.destinationDir property has been deprecated. This is scheduled to be removed in Gradle 8.0. Please use the destinationDirectory property instead. Consult the upgrading guide for further information: https://docs.gradle.org/7.3.3/userguide/upgrading_version_7.html#compile_task_wiring

Related search in [groovy-android-gradle-plugin](https://github.com/groovy/groovy-android-gradle-plugin/search?q=destinationDir)