dariuszseweryn/RxAndroidBle #752: Support Android 12 Bluetooth Permissions

mattinger, 2021-05-20:
With Android 12 comes two new bluetooth permissions that will replace Location Permission:

* Bluetooth Scan
* Bluetooth Connect

The library needs to be updated to reflect these states.  In addition, Bluetooth Connect is only
required if you're going to open a gatt connection to the the device, so it shouldn't block the
operation of the library.  Currently the RxBleClient.state is a progressive check that ends up with an isReadyForScanning
when all permissions and services have turned on.

I'd recommend that this be broken out into two states.  Once for scanning and one for connecting.

ArthurMytsko, 2021-06-10:
Is there any progress on that taks or estimates?
Android 12 Beta 2 was released yesterday, we are not far away from the final APIs (in August)

AndHain, 2021-07-08:
Are there any updates to this task?
Will the library be ready for the upcoming changes in Android 12?

hosek, 2021-07-09:
Btw there is also **ACCESS_BACKGROUND_LOCATION** missing for API 29+ (can't perform scan without it)
https://developer.android.com/about/versions/10/privacy/changes#app-access-device-location
Just adding note here https://github.com/Polidea/RxAndroidBle#permissions will be helpful

SaqibJDev, 2021-07-30:
Is there even a plan for this change? It would be really helpful not to force users to turn on location only for Bluetooth.

dariuszseweryn, 2021-07-30:
I have it on my plate. I am currently running this project on my free time which I do not have plenty unfortunately. 🤷‍♂️

mattinger, 2021-07-30:
> Btw there is also **ACCESS_BACKGROUND_LOCATION** missing for API 29+ (can't perform scan without it)
> https://developer.android.com/about/versions/10/privacy/changes#app-access-device-location
> Just adding note here https://github.com/Polidea/RxAndroidBle#permissions will be helpful

@hosek You don't need background location to scan while the app is active.  I agree that if you want to build an app that's scanning while the app is backgrounded you'll need it.  But that should only be necessary under certain circumstances where you're maybe looking to recognize being in range of certain devices.  But in general, you don't need it.

mattinger, 2021-07-30:
> Is there any info whether on Android 12 / API 31 scan results will get returned if Location Services are turned off while the scan is running?
> I do not have access to any device that can run Android 12.

That's a great question.  Usually if you're targeting an older SDK (in this case Android 11/Api 30), you'd fallback to the previous behavior, or something compatible to that.  I don't see anything in particular about this scenario though in the feature docs: https://developer.android.com/about/versions/12/features/bluetooth-permissions

joaquindrop, 2021-08-20:
> Is there any info whether on Android 12 / API 31 scan results will get returned if Location Services are turned off while the scan is running?
> I do not have access to any device that can run Android 12.

Hi @dariuszseweryn, I have access to a Pixel 5 running the current Beta of Android 12. I'm interested in RxAndroidBle supporting API 31, I wouldn't mind testing anything and provide logs that could help to close this issue.

dariuszseweryn, 2021-08-20:
Awesome, thank you for the proposition. I've got some support from a Google employee and I _think_ I have all needed knowledge about how the new OS is working.
I have now a metric tone of things to manage in my private life and it is hard for me to schedule work. When I will have something workable I will share it. Plus if anyone would like to get involved in writing the actual code / helping me — I can draw out a set of tasks to do and at least some hints on how to achieve them.

mattinger, 2021-08-20:
I will check with the appropriate people at work if i can contribute (have to go through a whole process).  But there's definitely some other suggestions of things i'd like to support (like ble encryption during gatt connect via a pre-computed pin).  That's definitely a big need i have coming down the pipe and i don't see anything in the library that supports it.

remonh87, 2021-09-23:
@mattinger  did you already work on this issue? Would be great to have this functionality in. We also need this for our flutter library. Let me know if I can help out.

sung2063, 2021-10-05:
Any progress on Android 12 Bluetooth permissions?

dariuszseweryn, 2021-10-06:
I have just pushed `1.13.0` — please check it out as I do not have capabilities myself.

mattinger, 2021-10-07:
> Originally this was targeting 29, but after updating the targetSDKVersion to 31, I'm now getting this error.
> 
> LOCATION_PERMISSION_NOT_GRANTED for rxBleClient.getState() before starting the scan.
> 
> I have confirmed that location permissions have been granted as my app also uses this for GPS Speed readings.
> 
> Edit: I've also added in the changes where you have to request 'ACCESS_FINE_LOCATION' along with 'ACCESS_COARSE_LOCATION' in a single runtime request when targeting Android 12.
> 
> I can confirm that my callbacks are showing that I have all the necessary permissions required for scanning.
> 
> Edit #2: My bad, I didn't realize we had to also request BLUETOOTH_SCAN and BLUETOOTH_CONNECT permissions manually after app-launch. After adding those in scanning is working again!

So are you saying if we want to scan, we have to request all 3 (BLUETOOTH_SCAN plus the two location permissions)?  I would think if set the neverForLocation you wouldn't need that.

mattinger, 2021-10-08:
@dariuszseweryn Just looking for some confirmation on what permissions it checks based on OS.
Assuming you set neverForLocation, would isReadyForScanning adapter to the OS?

For 30 and below: check fine location / coarse location
For 31 and higher: check bluetooth scan

Our app does both a scan and a connect, so we'll likely have to wait until your changes are ready for breaking the state into 2 pieces.  It's just unclear to me what the behavior is for the current version that's published.

remonh87, 2021-10-17:
I can confirm it works like intended many thanks! 🙏

techker, 2021-11-09:
is i possible to update the Sample-Kotlin for these fixes?Cause on OS12 it doesn't work.Thx

mattinger, 2022-03-03:
@dariuszseweryn Is there any plan to add functions around permission checks for gatt connections?  Maybe something like isPermissionRequiredToConnect and getRecommendedConnectRuntimePermissions

nrbrook, 2022-03-03:
@mattinger I am implementing this in https://github.com/dariuszseweryn/RxAndroidBle/pull/768

PR #762 (2021-10-06): WIP: Support API 31 (Android 12) new Bluetooth permissions
API 31 introduces new Bluetooth permissions required for BLE operations. In the context of RxAndroidBle the important ones are:
- `BLUETOOTH_SCAN` — replaces previously needed ACCESS_FINE_LOCATION
- `BLUETOOTH_CONNECT` — a new concept

Considerations:
The library's AndroidManifest.xml does include `ACCESS_COARSE_LOCATION` and `ACCESS_FINE_LOCATION` since API 23. Technically these could be restricted to `maxSdkVersion="30"` but it could be a breaking change for apps that require location access otherwise. There is a new example in the R [... cut]
PR #787 (2022-05-28): Added Android 12 connect permission helper functions
Related to #782 
Closes #752
closed 2022-05-28 by dariuszseweryn