dariuszseweryn/RxAndroidBle #753: Maybe the doc should explain how to install the library?

steph643, 2021-05-27:
I've just spent several hours installing RxAndroidBle into my project. The web is full of contradictory information about this. I think the README.md file should begin with a chapter called "Installation".

dariuszseweryn, 2021-05-27:
Could you elaborate what exactly you find confusing?

From what I know this library is no different in installing on a project than most of other, established, libraries. It has a readme which does describe steps and there are example projects as well.

steph643, 2021-05-28:
Hi @dariuszseweryn. I must be dumb, but I found neither a readme with installation steps, nor example projects with the proper `implementation` line (to import the lib from a repository). Would you please tell me where those are? This way I will try to understand how come I didn't see them.

dariuszseweryn, 2021-05-28:
If you use Gradle (which is the default for new projects started in Android Studio) then the proper dependency clause is [here in the Readme](https://github.com/Polidea/RxAndroidBle#gradle).

steph643, 2021-05-28:
That's it, I'm dumb indeed.
But notice that this is expected to be located at the top of the readme (as this is the first thing you need), not at the bottom. Example [here](https://github.com/ReactiveX/RxJava).

dariuszseweryn, 2021-05-28:
@steph643 — please check the PR if it is now better

steph643, 2021-05-28:
That looks much better to me, thanks.

PR #755 (2021-05-29): Promote Getting Started in Readme
Apparently installation instructions and permissions were not visible enough for newcomers.
Closes #753
closed 2021-05-28 by steph643