dariuszseweryn/RxAndroidBle #757: Should multiple subscriptions to the same scan observable start a new scan every time?

msmoljan, 2021-07-01:
**Is your feature request related to a problem? Please describe.**
I need to simultaneously observe scan results using two different subscriptions. Due to the user being able to shift back and forth between scanning and the connected state, the app makes two new scan subscriptions after every disconnection. 

It seems to me that if I subscribe to the same observable returned by `scanBleDevices(...)` multiple times (indicated by the javadoc as "safe"), I use up the amount of scanning I'm allowed to start pretty quickly before Android's throttling kicks in. I.e. RxAndroidBle tells Android to start scanning again, even though I haven't unsubscribed from the previous scan. 

In my case, I'm only allowed to jump back and forth between scanning and connection twice within 30s (the 2 scans are started 2 times, i.e. 4 subscriptions). The third time I try connecting, the 6th scan starts and the throttling makes my app unusable. 

**Describe the solution you'd like**
I'd expect the same observable instance to internally start BLE scanning only once, and new subscriptions to just "piggyback" on this result stream without initiating a new scan. As long as there are any subscriptions, the scanning is active, and the scanning stops when the last subscriber unsubscribes. 

In my case, this would mean that I'd be able to move between scanning and connected states 5 times within 30s instead of two, which isn't ideal but it would turn this critical issue into an edge case.

I was able to easily reproduce this issue in my project by making many (6+) subscriptions to the same scan observable at once. If this is not an expected behavior, please let me know and I'll also try it on a clean project. This way I can assure that there wasn't something else in my code causing it.  

Overall, I expected that the javadoc comment about being able to safely subscribe to the scan observable multiple times referred to avoiding the throttling issue, but I'm not sure if I misunderstood that or if this is indeed a library issue. Is starting a new scan every time an expected behavior of the library?

msmoljan, 2021-07-01:
Apart from that, thanks for the great work you're doing! :) 

Let me know if I can help out somehow.

dariuszseweryn, 2021-07-02:
This is the correct behaviour. The library tries not to do anything surprising. The Observable starts the scan on subscription and stops it on dispose.
You may want to check what RxJava `.share()` operator does as it seems to fit your use-case.

msmoljan, 2021-07-05:
Hi Dariusz, thanks a lot for the info and for the `.share()` tip! 

I'll close the issue then :+1:

closed 2021-07-05 by msmoljan