dariuszseweryn/RxAndroidBle #758: Uncaught exception: SingleResponseOperation.timeoutFallbackProcedure

sigurdurrafn, 2021-07-08:
An exception (SingleResponseOperation.timeoutFallbackProcedure, stack trace below) happens within RxAndroidBle causing our app to crash.

**To Reproduce**

This happens to some of our customers on Android 11 as they use our ble product.

**Expected behavior**

It's not known what causes the issue, but we expect the exception to be passed to our app, where we could take action there. Log the issue, disconnect etc.

**Smartphone :**
 - Device: A few from  OPPO, samsung, HUAWEI, Xiaomi 
 - OS: Android 11
 - Library version: 1.10.5

```
com.polidea.rxandroidble2.internal.SingleResponseOperation.timeoutFallbackProcedure (SingleResponseOperation.java:91)
com.polidea.rxandroidble2.internal.SingleResponseOperation.protectedRun (SingleResponseOperation.java:54)
com.polidea.rxandroidble2.internal.QueueOperation$1.subscribe (QueueOperation.java:41)
io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual (ObservableCreate.java:40)
io.reactivex.Observable.subscribe (Observable.java:12267)
io.reactivex.internal.operators.observable.ObservableUnsubscribeOn.subscribeActual (ObservableUnsubscribeOn.java:32)
io.reactivex.Observable.subscribe (Observable.java:12267)
com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1.run (FIFORunnableEntry.java:56)
io.reactivex.internal.schedulers.ScheduledDirectTask.call (ScheduledDirectTask.java:38)
io.reactivex.internal.schedulers.ScheduledDirectTask.call (ScheduledDirectTask.java:26)
java.util.concurrent.FutureTask.run (FutureTask.java:266)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
java.lang.Thread.run (Thread.java:923) 
```
```

dariuszseweryn, 2021-07-08:
The stack trace you included points to where the exception is being raised. Why do you think it is a library bug?

sigurdurrafn, 2021-07-09:
Great, I'll then catch the bug and close this issue.

All I see are internal rx classes, where is the exception being raised so I can catch it?

dariuszseweryn, 2021-07-09:
Do you have an error handler in each `.subscribe()` you wrote?
There is a slight chance for a race condition when disposing streams — in that case you can use `RxJavaPlugins.setErrorHandler()` (or similar) method

sigurdurrafn, 2021-07-09:
Yes we handle all thrown exceptions, this happens despite that. That's why we reported this as a bug.
 
Is this race condition tracked ? 
I see https://github.com/Polidea/RxAndroidBle/issues/694 but seems to be a different issue.

Will add a global exception handler in any case, thanks.

dariuszseweryn, 2021-07-09:
You haven't disclosed the source code that is expected to be handling the error nor you did not disclose the full stack trace and the exact exception message. There is #213 which is supposed to address the `UndeliverableException`. But if you have an uncaught exception then I would think that you lack an error handler in some `.subscribe()`

closed 2021-08-04 by dariuszseweryn