dariuszseweryn/RxAndroidBle #759: GATT_CONN_TERMINATE_LOCAL_HOST Error

tejma17, 2021-07-16:
I am getting the error 

```
com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:272)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:267)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:192)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
```

when I tried connecting to the device and reading the name from the characteristic UUID.

This is my code:

```
device = rxBleClient.getBleDevice(address);
        UUID batteryServiceUuid = UUID.fromString("00002A19-0000-1000-8000-00805F9B34FB");
        connectDisposable = device.establishConnection(false)
                .flatMapSingle(rxBleConnection -> rxBleConnection.readCharacteristic(batteryServiceUuid))
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe(
                        characteristicValue -> {
                            Log.d("BATTERY", "SUBS");
                            showErrorText(new String(characteristicValue));
                        },
                        throwable -> {
                            Log.d("BATTERY", "ERROR");
                            showErrorText(throwable.getLocalizedMessage()+" "+throwable.getStackTrace()[0]);

                        }
                );
````

Please help.

dariuszseweryn, 2021-07-17:
The library correctly delivers the error the Android API provides. Why do you think it is a bug?

tejma17, 2021-07-17:
Yeah. I mistakenly added the label bug. Extremely sorry for that. I just
need help with this error to get started with read write. Thank you.

dariuszseweryn, 2021-07-17:
You may help me and yourself by conforming to the issue template you ignored. Most probably though the problem lies on the peripheral side which does something unexpected for the Android BLE stack.

tejma17, 2021-07-18:
The issue arises when I want to read a characteristic. Scanning and connecting to the device works properly. 

Steps to reproduce:
1. I have scanned available devices and connected to one of them.
2. Then I tried to read the battery status using the read function provided in the readme. 
3. On the first run, it's throwing the error:
```
com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:272)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:267)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:192)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
```

4. When I tried to read again, the error is, `Already connected to MAC='XX:XX:XX:XX:XX:XX' `

I am using two android mobile devices here running on Android 10.

dariuszseweryn, 2021-07-18:
Your description does not add much context. Please check the issue template and provide info according to it

closed 2021-08-04 by dariuszseweryn