dariuszseweryn/RxAndroidBle #772: Android 12: Duplicate declarations of android.permission.ACCESS_FINE_LOCATION w different maxSdkVersions.

JimTompkins, 2021-12-13:
**Describe the bug**
App works as expected on phone, but I get an error when uploading the appbundle to the Play store: Duplicate declarations of permission android.permission.ACCESS_FINE_LOCATION with different maxSdkVersions. 

**To Reproduce**
Not sure what change made this start happening... I updated the sdk version to 31.

**Expected behavior**
Should not give error upon upload.

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S7
 - OS: Android 8.0.0
 - Library version: 5.0.0

**Additional context**
I copied in the permissions from the AndroidManifest.xml in the example. 
`    
    <uses-permission
            android:name="android.permission.BLUETOOTH"
            android:maxSdkVersion="30" />
    <uses-permission
            android:name="android.permission.BLUETOOTH_ADMIN"
            android:maxSdkVersion="30" />

    
    <uses-permission-sdk-23
            android:name="android.permission.ACCESS_COARSE_LOCATION"
            android:maxSdkVersion="30" />
    <uses-permission-sdk-23
            android:name="android.permission.ACCESS_FINE_LOCATION"
            android:maxSdkVersion="30" />

    
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission
            android:name="android.permission.BLUETOOTH_SCAN"
            android:usesPermissionFlags="neverForLocation" />` 

I do a find-in-files for "ACCESS_FINE_LOCATION" and it only appears in the AndroidManifest.xml as expected.

I've done a clean and invalidate caches and restart.

Flutter doctor looks ok (to me):
`Doctor summary (to see all details, run flutter doctor -v):
[√] Flutter (Channel master, 2.6.0-12.0.pre.1010, on Microsoft Windows [Version 10.0.19043.1348], locale en-CA)
[!] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    X Android license status unknown.
      Run `flutter doctor --android-licenses` to accept the SDK licenses.
      See https://flutter.dev/docs/get-started/install/windows#android-setup for more details.
[√] Chrome - develop for the web
[√] Android Studio (version 2020.3)
[√] Connected device (3 available)

! Doctor found issues in 1 category.`

Flutter analyze is clean.

Thanks for any hints.

dariuszseweryn, 2021-12-14:
What version of the lib you use?

JimTompkins, 2021-12-14:
5.0.0

JimTompkins, 2021-12-14:
I found my problem.  I was including another library by mistake.  Backstory: flutter_reactive_ble doesn't work for my app for iOS so I use a different package, and have to change back and forth...

dariuszseweryn, 2021-12-14:
I am glad that you found your problem. This is not a flutter library repository though :)
Cheers