dariuszseweryn/RxAndroidBle #773: Support secondary services

teddyvn, 2021-12-30:
It is seem as RxAndroidBle hasn't support secondary service which is included in a service.
In class  RxBleDeviceServices, method  getService, method  getCharacteristic, we just iterate through primary services. We didn't get the secondary services from the primary services
Could you consider to support this?
Thanks

dariuszseweryn, 2022-05-28:
Hello,
I was considering this but I could not lay my hands on any actual/production BLE device that used this feature. Without a role-model it is hard for me to proof an API. Do you know of any device that could help?

teddyvn, 2022-06-30:
Oh about  the real device, I didn't know. I left this suggest when I developed a simulator for my project. I dug into the API then I see that it has a secondary service.

dariuszseweryn, 2022-06-30:
Yeah, I have seen the API but I have not encountered a real life example

dariuszseweryn, 2022-06-30:
Alternative question: what device/SDK have you used?

teddyvn, 2022-07-01:
These are gate locks (many kinds). It is a smart lock. Sorry I can't tell it name here.