dariuszseweryn/RxAndroidBle #783: rxBleClient.scanBleDevices(...) Always Returns Throwable and Not ScanResult

EddieXu123, 2022-04-21:
**Describe the bug**
I am writing an Android App to try and discover nearby bluetooth devices. I copied the instructions in the Readme up until the "Device Discovery" portion and everything seems to be working (this.startActivityForResult(enableBtIntent, REQUEST_ENABLE_BT); makes the app ask the user to turn on Bluetooth). However, I am copying the next portion and it continues to go into the throwable part.

**To Reproduce**
This is my code (simplified without the UI stuff):

```
@Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        
        
        
        
        

```

**Expected behavior**
For now, I'm just hoping to be able to access the particular bluetooth low energy device in the code. Some advice on how to best grab that would also be greatly appreciated!

**Smartphone (please complete the following information):**
 - Device: Google Pixel 5
 - OS: Android 12
 - Library version: 1.13.1

**Additional context**
For now this app can be simplified to putting the Readme code in my onCreate() function at the beginning. Also, I'm not sure if the "context" I am using is correct - I'm pretty new to Android development so if this isn't correct, I'd appreciate the info! Thank you so much for your help!

EddieXu123, 2022-04-21:
Also, sorry this is definitely not a bug with your library but with my code/setup, but I wasn't sure how to file an issue otherwise

ZacharyTech, 2022-04-26:
I haven't seen the author's dynamic for a long time. I [forked](https://github.com/ZacharyTech/RxAndroidBle) the codebase and made some changes to keep it working, hope it helps

tolot27, 2022-04-26:
@ZacharyTech Maybe you can contribute your changes to https://github.com/NRB-Tech/RxAndroidBle? This repository seem to be more active.

EddieXu123, 2022-04-26:
Hi I actually figured out how to get the bluetooth scan working (It was with enabling location services on my phone haha)

I however, have one more error which I think has to do with writing/reading too much to the device. I am writing to my device and it works for ~2-3 writes, but then I get an `com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC=...` error. I think I've seen it before but am wondering if there are any quick fixes haha (I think it's because I am writing to the bluetooth device a lot of times (with a listener)). Not sure if this can be worked on

ZacharyTech, 2022-04-27:
> 

Great idea, I'll try this

ZacharyTech, 2022-04-27:
> @ZacharyTech Maybe you can contribute your changes to https://github.com/NRB-Tech/RxAndroidBle? This repository seem to be more active.

Great idea, I'll try this

ZacharyTech, 2022-04-27:
> Hi I actually figured out how to get the bluetooth scan working (It was with enabling location services on my phone haha)
>
> I however, have one more error which I think has to do with writing/reading [...]

please provide the code

EddieXu123, 2022-04-27:
Sure, thank you! Here is the seekbar that is writing to the bluetooth device through a seekbar onChangeListener:

```
    public void createSeekBar() { // Called in onCreate
        { ... Instantiation ...}
        brightness_seekbar.setOnColorChangeListener((colorBarPosition, alphaBarPosition, color) -> {
            if (isNear(colorBarPosition, brightness_position)) {
                return;
            }

            brightness_position = colorBarPosition;
            byte[] byteToChange = getInputByte(colorBarPosition);
            writeToLamp("brightness", byteToChange);
            rectangle.setColor(saturation_seekbar.getColor());
            rectangle.setColorFilter(applyLightness(100 - brightness_position));
        });

    /* Getting the Device */
    private RxBleDevice getMyLampi(String lampiMAC) {
        RxBleClient rxBleClient = RxBleClient.create(this);

        return rxBleClient.getBleDevice(lampiMAC);
    }

    /* Function to write to device (lamp) */
    private void writeToLamp(String bar, byte[] bytesToWrite) {
        RxBleDevice myLampi = getMyLampi(lampiMacAddress);
        UUID toWriteUUID;

        // Write to the Lamp
        temp = myLampi.establishConnection(false)
                .flatMapSingle(rxBleConnection -> rxBleConnection.writeCharacteristic(toWriteUUID, bytesToWrite))
                .subscribe(
                        characteristicValue -> {
                            // Characteristic value confirmed.
                            Log.d("Result", "CHANGED: " + Arrays.toString(characteristicValue));
                        },
                        throwable -> {
                            // Handle an error here.
                            Log.d("Result", "ERROR : " + throwable.toString());
                        }
                );
    }

```

I believe the issue is that I'm calling the getMyLampi function every time it's writing, which sends a bunch of calls to get the same device every time, which maybe overloads the library? I tried creating a global myLampi to represent my device and only calling the getMyLampi function in the onCreate and then writing to the lamp through the myLampi.establishConnection(false)... line, but this gives a "BleAlreadyConnectedException". I looked at the sample code but was having some trouble reproducing it with my library. For now doing a few non-rapid writes/reads to the device work with my current code, but help to reduce the number of calls (maybe this is the error?) would definitely be appreciated! Thank you :)

PS, if further code is needed, I basically only have one file in my Lamp project: https://github.com/EddieXu123/LampiAndroid/blob/main/app/src/main/java/com/example/myfirstapp/MainActivity.java

Thank you so much for all your help!

ZacharyTech, 2022-04-29:
Follow `myLampi.establishConnection(false)` , both complete and error will disconnect the connection

dariuszseweryn, 2022-05-28:
Hello,
For the initial problem – it is always good to check the entire message of the exception. I tried to make them obvious for troubleshooting. For the latter your exception lacked the most interesting part – that is the error status.
I hope you are no longer seeing the issue — if not please provide more info and I will reopen.
 
I plan to continue here but https://github.com/NRB-Tech/RxAndroidBle seems to be a solid alternative. The author there did a lot of great work 💪

closed 2022-05-28 by dariuszseweryn