dariuszseweryn/RxAndroidBle #788: Investigate: Android 12, BLUETOOTH_CONNECT & getBondedDevices()

dariuszseweryn, 2022-05-28:
As hinted in #768 — on Android 12 there apparently is a need to have BLUETOOTH_CONNECT permission when calling RxBleClient.getBondedDevices().

If you, the reader, have an Android 12 device and you're willing to help and check — please do.

If that is the case update Javadocs.

DaBs, 2022-06-15:
Hi Darius, I made the original PR that I can see you've merged now, and that seems to be entirely correct based on what we can see of usage.

How should we go about it from here?

dariuszseweryn, 2022-06-27:
@DaBs I am thinking whether there should be a separate library exception or is the system one just fine, especially that this API is synchronous

DaBs, 2022-07-08:
@dariuszseweryn I think system error should be fine honestly - getting the error is clear and concise.

dariuszseweryn, 2022-07-09:
I have added that info to Javadoc and left the feature as it is. Thanks for the opinion!

PR #804 (2022-07-09): Updated RxBleClient#getBondedDevices() javadoc.
Mentioning new requirements for Android 12 / API >= 31.

Closes #788
closed 2022-07-09 by dariuszseweryn