dariuszseweryn/RxAndroidBle #789: 2 files found with path 'META-INF/com.google.dagger_dagger.version'.

atick-faisal, 2022-06-01:
After updating to 1.14.0 from 1.10.2 getting this error during the build:
`2 files found with path 'META-INF/com.google.dagger_dagger.version'.`

dariuszseweryn, 2022-06-03:
Looking into it

dariuszseweryn, 2022-06-03:
Hey @atick-faisal — could you check the `1.14.1-SNAPSHOT` release if it fixes your problem?

atick-faisal, 2022-06-03:
Hello @dariuszseweryn. Thanks for the reply. I added the following in my build.gradle file:
`implementation("com.polidea.rxandroidble2:rxandroidble:1.14.1-SNAPSHOT")`

Getting this error:
```
> Could not resolve com.polidea.rxandroidble2:rxandroidble:1.14.1-SNAPSHOT.
         > Unable to load Maven meta-data from https://jitpack.io/com/polidea/rxandroidble2/rxandroidble/1.14.1-SNAPSHOT/maven-metadata.xml.
            > Could not GET 'https://jitpack.io/com/polidea/rxandroidble2/rxandroidble/1.14.1-SNAPSHOT/maven-metadata.xml'.
               > Read timed out
```

Am I doing something wrong?

dariuszseweryn, 2022-06-04:
[Instructions how to use snapshot releases are in the README](https://github.com/dariuszseweryn/RxAndroidBle#snapshot)

atick-faisal, 2022-06-04:
Thank you @dariuszseweryn. This solved the issue. Build completed successfully.

dariuszseweryn, 2022-06-04:
Awesome, thank you for verifying

dariuszseweryn, 2022-06-14:
1.15.1-SNAPSHOT should be OK. Regular 1.15.1 should be available shortly

PR #790 (2022-06-04): Fixed build for apps using Dagger
Dagger in 2.25.4 introduced a tracking file ('META-INF/com.google.dagger_dagger.version') which is published with the archive. If the file is in a library and the app also uses Dagger it can break a build by default because the file is already there.

Fixes #789
PR #794 (2022-06-14): Fixed build for apps using Dagger (regression)
Fixes #789