dariuszseweryn/RxAndroidBle #792: Unable to connect Android sample app with the peripheral

ashutosh7107, 2022-06-08:
Here are the logs when I am trying to connect

E/InsulinTransmitter: Cannot write client characteristic config descriptor (code 3) with characteristic UUID 00002a52-0000-1000-8000-00805f9b34fb
2022-06-02 17:22:40.050 7305-9029/com.lilly.dh.myTempoLogbook E/LillyBleDevice: error
com.polidea.rxandroidble2.exceptions.BleCannotSetCharacteristicNotificationException: Cannot write client characteristic config descriptor (code 3) with characteristic UUID 00002a52-0000-1000-8000-00805f9b34fb
at com.polidea.rxandroidble2.internal.connection.NotificationAndIndicationManager$7.apply(NotificationAndIndicationManager.java:241)
at com.polidea.rxandroidble2.internal.connection.NotificationAndIndicationManager$7.apply(NotificationAndIndicationManager.java:238)
at io.reactivex.internal.operators.completable.CompletableResumeNext$ResumeNextObserver.onError(CompletableResumeNext.java:82)
at io.reactivex.internal.operators.observable.ObservableIgnoreElementsCompletable$IgnoreObservable.onError(ObservableIgnoreElementsCompletable.java:61)
at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1$1.onError(FIFORunnableEntry.java:73)
at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:70)
at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
at com.polidea.rxandroidble2.internal.util.QueueReleasingEmitterWrapper.onError(QueueReleasingEmitterWrapper.java:45)
at io.reactivex.internal.observers.DeferredScalarDisposable.error(DeferredScalarDisposable.java:100)
at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onError(SingleToObservable.java:77)
at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver.onError(SingleTimeout.java:150)
at io.reactivex.internal.operators.single.SingleMap$MapSingleObserver.onError(SingleMap.java:69)
at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java:102)
at io.reactivex.internal.observers.BasicFuseableObserver.onError(BasicFuseableObserver.java:100)
at io.reactivex.observers.SerializedObserver.onError(SerializedObserver.java:153)
at io.reactivex.internal.operators.observable.ObservableDelay$DelayObserver$OnError.run(ObservableDelay.java:128)
at io.reactivex.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:66)
at io.reactivex.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:57)
at java.util.concurrent.FutureTask.run(FutureTask.java:266)
at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
at java.lang.Thread.run(Thread.java:919)
Caused by: com.polidea.rxandroidble2.exceptions.BleGattDescriptorException: GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 253 (GATT_CCC_CFG_ERR), type BleGattOperation{description='DESCRIPTOR_WRITE'}. (Look up status 0xfd here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)
at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:259)
at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onDescriptorWrite(RxBleGattCallback.java:174)
at android.bluetooth.BluetoothGatt$1$10.run(BluetoothGatt.java:580)
at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
at android.bluetooth.BluetoothGatt$1.onDescriptorWrite(BluetoothGatt.java:575)
at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:294)
at android.os.Binder.execTransactInternal(Binder.java:1021)
at android.os.Binder.execTransact(Binder.java:994)

Any idea?

dariuszseweryn, 2022-06-08:
Yup, got one idea but I would like to first ask:
1. What phone you use?
2. What OS does the phone have?
3. What do you think is going on judging by the stack trace?

ashutosh7107, 2022-06-08:
I have used three phones  1. Xiaomi POCO F1 - Android 10 
2. Samsung M31s - Android 11
3. Samsung S10 - Android 12

ashutosh7107, 2022-06-08:
Yes, I checked. I was getting no logs or errors in peripheral side also the peripheral app is working with all other applications so this makes me think that might be some issue in this library or somewhere I am doing wrong in implementing this library. Not sure at all, got stuck in this since 3 days

dariuszseweryn, 2022-06-08:
The error status is provided by the OS. That is most probably because the peripheral is not up to BLE spec.
If that is the case — there are alternative modes for setting notifications.

ashutosh7107, 2022-06-08:
Ok, so if that is the case. What should be our approach? What are the alternative modes for setting notifications? Could you please guide me on this

ashutosh7107, 2022-06-08:
Also sometimes If I am not getting logs, it is going to connecting state and quickly disconnected where I am getting connection timeout popup

dariuszseweryn, 2022-06-09:
> What should be our approach?

You may verify if your fake peripheral is correct in terms of BLE specification or you may try different things (like the modes) and see what will stick
> What are the alternative modes for setting notifications?

RxBleConnection#setupNotification(UUID, NotificationSetupMode) – Javadocs should give you enough background
> Could you please guide me on this

I am already doing that – I will just not dive into how the Raspberry Pi peripheral is written as this is typically a rabbit hole. If several implementations of Android BLE stack give you status 253 then it is most probably problem with the peripheral's implementation

ashutosh7107, 2022-06-13:
Hey, Thank you for your suggestion. 

I have analysed both side peripheral and android central app and found there was issue in central android app. It was failing for the notify characteristics while setting notifications. Now two things are happening here let me explain you. 

There are two peripheral app:
1. iOS peripheral app, 
2. Web app written in python using raspberry pi(acting as peripheral).

Earlier it was "setupIndication" for the notify characteristics and it was working fine with android central app but not with Web app, but as I changed this to "setupNotification" it started working with web peripheral app and getting error in iOS peripheral app. 

So, just wanted to check with you is this some kind of limitation or is there any other way also to fix this which can be work with both peripherals.

Thanks in advance..!!

dariuszseweryn, 2022-06-13:
The characteristic you want to have notifications/indications from needs to support them. Check its properties. If you will set debug logs on the RxAndroidBle you would get a list of characteristics with their properties. I bet that on connecting to iOS you would get a different result than when connecting to RPi.

dariuszseweryn, 2022-06-20:
Please set DEBUG log level on the client and paste connection logs for both RPi and iOS – there should be enough info there to diagnose the problem