dariuszseweryn/RxAndroidBle #796: Can you provide a disconnection function?

advmeds-service, 2022-06-23:
I encountered a situation
When I connect 2 devices at once
When the connection is successful, the notification subcribe is listened to
When one is off and the other is in use
The program keeps trying to connect the two devices every 5 seconds
but because the connection is not disconnected
so it has to wait until the one that is switched off GATT ERROR 
The one that is in use will not successfully connect to receive data
This is a bit of a long wait.

I want to disconnect manually after receiving the data
So that the next time I connect, I don't have to wait for GATT ERROR.
I don't know if this will improve the long waiting time of the one in use.

thank you

dariuszseweryn, 2022-06-23:
RxAndroidBle was designed to not have any state management (i.e. disconnection function) — when the user is no longer interested in keeping the connection they just dispose the flow.
You should be able to dispose your connection when you receive the data. You can post your issue on stack overflow and tag it with `rxandroidble` and `rxjava` and there is a good chance someone will help you fix your code.

advmeds-service, 2022-06-27:
ok thank you

closed 2022-06-27 by advmeds-service