dariuszseweryn/RxAndroidBle #797: Got issue when `setCharacteristicNotification()` run into an infinite loop

lng-dawn, 2022-06-26:
**Describe the bug**
I have tried to fix issue related to ble connection. I have 2 test devices, 1 of them has a special firmware (it will disconnect right after having connected) and the purpose of this is to test the edge case when user just stay in the boundary between connect and disconnect.

**To Reproduce**
Connect 2 devices on app

- 1 device has normal firmware
- another device has special firmware to connect and disconnect continuously
- run the app after around 10 mins and look at behaviour of 2 devices:

1. the normal device firmware act normal and has a stable connection
2. the special device firmware act has expected (connect and disconnect), after 10 mins, it just stuck in a limbo state. And there is a log `D/BluetoothGatt: setCharacteristicNotification() - uuid: e2fa0108-b177-4e96-a550-4a47e40adcb7 enable: false` running in a forever loop

**Expected behavior**
Expect to see the special firmware connect/disconnect continuously and not being in a stuck state. Also handle case `setCharacteristicNotification() ` forever loop

**Smartphone (please complete the following information):**
 - Samsung S21
 - Library version: 1.15.0

**Logs from the application when bug occurs:
```
BluetoothGatt: setCharacteristicNotification() - uuid: e2fa0108-b177-4e96-a550-4a47e40adcb7 enable: false
```

**Additional context**
This is being logged forever

dariuszseweryn, 2022-06-26:
Hi, thanks for sharing but this information is not actionable. The single log line you pasted is a normal part of disabling a notification and does not provide much info. The situation you describe has not surfaced to anyone during past 6 years of library lifecycle so most probably you have a wrong implementation of the connection flow in your app (which you had not disclosed).

lng-dawn, 2022-06-26:
@dariuszseweryn really appreciate for your feedback. Basically, I'm having issue with BLE connection. Sometime I got error `BleAdreadyConnectedException`. And this is the code for establishing connection
```
fun connectRxBleDevice(rxBleDevice: RxBleDevice, mtuSize: Int = 185): Observable<RxBleConnection> {
    return BleDeviceBondingManager.createBond(context, rxBleDevice)
      .filter { rxBleDevice.bluetoothDevice.bondState == BluetoothDevice.BOND_BONDED }
      .flatMap { device ->
        device.establishConnection(false)
          // this delay is added in order to get the system to properly connect and update services properly (will not work without this delay)
          .delay(2, TimeUnit.SECONDS)
          .flatMap { rxBleConnection ->
            rxBleDeviceConnectionStateObservable(rxBleDevice)
              .filter { it == RxBleConnection.RxBleConnectionState.CONNECTED }
              .flatMapSingle { rxBleConnection.requestMtu(mtuSize) }
              .map { rxBleConnection }
          }
      }
  }
```

And now  I try to follow the tutorial - `Tutorial: Connection Observable sharing` - https://github.com/dariuszseweryn/RxAndroidBle/wiki/Tutorial:-Connection-Observable-sharing, experiment option 3 which is creating `BleSessionManager `. Then function `connectRxBleConnection` becomes:

```
private val mapSessionManager = mutableMapOf<String, BleSessionManager>()

fun connectRxBleDevice(rxBleDevice: RxBleDevice, mtuSize: Int = 185): Observable<RxBleConnection> {
    return BleDeviceBondingManager.createBond(context, rxBleDevice)
      .filter { rxBleDevice.bluetoothDevice.bondState == BluetoothDevice.BOND_BONDED }
      .flatMap { device ->
        val session = if (mapSessionManager[device.macAddress] != null) {
          mapSessionManager[device.macAddress]
        } else {
          mapSessionManager[device.macAddress] = BleSessionManager(device)
          mapSessionManager[device.macAddress]?.setupConnection()
          mapSessionManager[device.macAddress]
        }

        session?.let {
          session.connection
            .flatMap { rxBleConnection ->
              rxBleDeviceConnectionStateObservable(rxBleDevice)
                .filter { it == RxBleConnection.RxBleConnectionState.CONNECTED }
                .flatMapSingle { rxBleConnection.requestMtu(mtuSize) }
                .map { rxBleConnection }
            }
            .doOnError { error ->
              if (error is BleAlreadyConnectedException || error is BleDisconnectedException) {
                Log.e("BluetoothHandler", "Got error $error, need to remove connection")
                //mapRxBleConnection.remove(rxBleDevice.macAddress)
                session.disconnect()
                session.setupConnection()
              }
              throw error
            }
        }
      }
  }
```

I haven't seen any error `BleAlreadyConnectedException ` when I use `BleSessionManager `. However, according to your comment, which means I have wrong implementation of connection flow. And it keeps show this log `BluetoothGatt: setCharacteristicNotification() - uuid: e2fa0108-b177-4e96-a550-4a47e40adcb7 enable: false`. Note that the function `connectRxBleDevice ` will be called whenever I detect that device lost connection (the check should be every 12s)

dariuszseweryn, 2022-06-26:
`establishConnection` does exactly this — establishes a connection — emits a single connection and maybe an error. I see you use `BleDeviceBondingManager.createBond(context, rxBleDevice)` which I do not know — you may try to check how often it emits, perhaps it is emitting more often than you think. The library by itself will not fall into a loop

lng-dawn, 2022-06-27:
Yeah, I doubled check fun `createBond`, it's looking fine

```
fun createBond(context: Context, rxBleDevice: RxBleDevice): Observable<RxBleDevice> {
    return Observable.create { emitter ->
      when (rxBleDevice.bluetoothDevice.bondState) {
        BluetoothDevice.BOND_BONDED -> {
          Log.e("BleDeviceBondingManager", "BLE device BONDED")
          emitter.onNext(rxBleDevice)
        }
        else -> {
          val receiver = object : BroadcastReceiver() {
            override fun onReceive(context: Context, intent: Intent) {
              Log.e("BleDeviceBondingManager", "receiving bonding change")
              val deviceBeingPaired = intent.getParcelableExtra<BluetoothDevice>(BluetoothDevice.EXTRA_DEVICE)

              if (deviceBeingPaired?.address == rxBleDevice.bluetoothDevice.address) {
                val state = intent.getIntExtra(BluetoothDevice.EXTRA_BOND_STATE, BluetoothDevice.BOND_NONE)
                val prevState = intent.getIntExtra(BluetoothDevice.EXTRA_PREVIOUS_BOND_STATE, BluetoothDevice.ERROR)

                when {
                  state == BluetoothDevice.BOND_BONDED -> {
                    Log.e("BleDeviceBondingManager", "BLE device BONDED")
                    context.unregisterReceiver(this)
                    emitter.onNext(rxBleDevice)
                  }
                  state == BluetoothDevice.BOND_BONDING -> {
                    Log.e("BleDeviceBondingManager", "BLE device BONDING")
                    /* no-op */
                  }
                  state == BluetoothDevice.BOND_NONE && prevState == BluetoothDevice.BOND_BONDING -> {
                    Log.e("BleDeviceBondingManager", "BLE device LOST BONDING")
                    emitter.onError(BleBondingFailedException())
                  }
                }
              }
            }
          }

          context.registerReceiver(receiver, IntentFilter(BluetoothDevice.ACTION_BOND_STATE_CHANGED))
          rxBleDevice.bluetoothDevice.setPin("000000".toByteArray(Charset.forName("UTF-8")))

          val createBondResult = rxBleDevice.bluetoothDevice.createBond()
          if (!createBondResult) {
            emitter.onError(BleBondingFailedException())
          }
        }
      }
    }
  }
```

I notice this log `D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: false` is being called lot (**with different uuid**) after BLE device disconnect right? However, when it runs into an infinite loop, the **uuid is just the same**, so it tried to disconnect but couldn't do that right?

One more 1 question about this function
```
fun rxBleDeviceConnectionStateObservable(rxBleDevice: RxBleDevice): Observable<RxBleConnection.RxBleConnectionState> {
    return rxBleDevice
      .observeConnectionStateChanges()
      .startWith(rxBleDevice.connectionState)
  }
```
Could it return wrong state sometime? Cause I saw my device is disconnected, but it still returned either `CONNECTED` or `CONNECTING` 
Please help to check, thank you very much

dariuszseweryn, 2022-06-27:
> I notice this log `D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: false` is being called lot (with different uuid) after BLE device disconnect right?

I do not know what you observe as you did not included fill logs. This line may appear _after_ the device is disconnected as it changes internal Android BLE stack property.

> However, when it runs into an infinite loop, the uuid is just the same, so it tried to disconnect but couldn't do that right?

AFAIK wrong — you are already disconnected yet the code tries to setup notification anyway

> Could it return wrong state sometime? Cause I saw my device is disconnected, but it still returned either `CONNECTED` or `CONNECTING`

No, this should not emit wrong state — it is tested. This may indicate that your code tries to connect when you do not expect it to.

lng-dawn, 2022-06-27:
Thank you for your reply. I think I solved the problem. 
Caching BLE connection with `BleSessionManager ` works pretty good. I haven't seen any `BleAlreadyConnectedException ` anymore :). 
The loop of trying to setup notification has been solved too.

dariuszseweryn, 2022-06-29:
Thank you for sharing. I close this issue.