dariuszseweryn/RxAndroidBle #798: Android 12 running java example (sample)

pascalpm, 2022-06-29:
Anyone found a solution to this? Running the java sample in Android 12 results to (debug mode):
 
**java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource ......**

Older Android versions do not ask for this permission and seem to work fine!

R1.15.1
Android Studio Chipmunk 2021.2.1 Patch 1
Sdk 32

dariuszseweryn, 2022-06-29:
Could you describe in more detail what steps did you take that ended up with such result?

pascalpm, 2022-06-30:
Here it is (from the branch "fix/permissions_logging". Hopefully this helps.

06/30 18:28:26: Launching 'sample' on samsung SM-A315G.
Install successfully finished in 543 ms.
$ adb shell am start -n "com.polidea.rxandroidble.sample/com.polidea.rxandroidble2.sample.example1_scanning.ScanActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER -D
Waiting for application to come online: com.polidea.rxandroidble.sample
Connecting to com.polidea.rxandroidble.sample
Connected to the target VM, address: 'localhost:51145', transport: 'socket'
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/droidble.sampl: Late-enabling -Xcheck:jni
E/USNET: USNET: appName: com.polidea.rxandroidble.sample
D/ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
D/ActivityThread: setConscryptValidator
D/ActivityThread: setConscryptValidator - put
V/studio.deploy: Startup agent attached to VM
V/studio.deploy: No existing instrumentation found. Loading instrumentation from instruments-962d6e1a.jar
W/droidble.sampl: DexFile /data/data/com.polidea.rxandroidble.sample/code_cache/.studio/instruments-962d6e1a.jar is in boot class path but is not in a known location
V/studio.deploy: ModifyParameterTransform failed: Landroid/app/ApplicationLoaders;
W/droidble.sampl: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
W/droidble.sampl: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
W/droidble.sampl: Redefining intrinsic method java.lang.Thread java.lang.Thread.currentThread(). This may cause the unexpected use of the original definition of java.lang.Thread java.lang.Thread.currentThread()in methods that have already been compiled.
W/droidble.sampl: Redefining intrinsic method boolean java.lang.Thread.interrupted(). This may cause the unexpected use of the original definition of boolean java.lang.Thread.interrupted()in methods that have already been compiled.
W/droidble.sampl: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
W/droidble.sampl: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
W/droidble.sampl: Current dex file has more than one class in it. Calling RetransformClasses on this class might fail if no transformations are applied to it!
D/ActivityThread: handleBindApplication()++ app=com.polidea.rxandroidble.sample
D/CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10481; state: ENABLED
W/ActivityThread: Application com.polidea.rxandroidble.sample is waiting for the debugger on port 8100...
I/System.out: Sending WAIT chunk
I/System.out: Debugger has connected
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
W/System: A resource failed to call close. 
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: waiting for debugger to settle...
I/System.out: debugger has settled (1484)
W/ActivityThread: Slow operation: 2021ms so far, now at handleBindApplication: Before HardwareRenderer
W/ActivityThread: Slow operation: 2046ms so far, now at handleBindApplication: After HardwareRenderer
D/ApplicationLoaders: Returning zygote-cached class loader: /system/framework/android.test.base.jar
V/GraphicsEnvironment: ANGLE Developer option for 'com.polidea.rxandroidble.sample' set to: 'default'
V/GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
D/LoadedApk: LoadedApk::makeApplication() appContext.mOpPackageName=com.polidea.rxandroidble.sample appContext.mBasePackageName=com.polidea.rxandroidble.sample
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/ActivityThread: handleBindApplication() --
I/BluetoothAdapter: BluetoothAdapter() : com.polidea.rxandroidble.sample
I/BluetoothAdapter: BluetoothAdapter() : com.polidea.rxandroidble.sample
D/OpenGLRenderer: RenderThread::requireGlContext()
D/hw-ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
E/OpenGLRenderer: Device claims wide gamut support, cannot find matching config, error = EGL_SUCCESS
W/OpenGLRenderer: Failed to initialize 101010-2 format, error = EGL_SUCCESS
D/OpenGLRenderer: RenderThread::setGrContext()
I/DecorView: [INFO] isPopOver=false, config=true
I/DecorView: updateCaptionType >> DecorView@2cf52e5[], isFloating=false, isApplication=true, hasWindowControllerCallback=true, hasWindowDecorCaption=false
D/DecorView: setCaptionType = 0, this = DecorView@2cf52e5[]
I/DecorView: getCurrentDensityDpi: from real metrics. densityDpi=420 msg=resources_loaded
W/droidble.sampl: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (unsupported, reflection, allowed)
[... cut ...]
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(205963980) in 48 ms
I/RxBle#CancellableDisposable: Scan operation is requested to stop.
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: Stop Scan with callback
W/ScanActivity: Scan failed
    java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource { uid = 10481, packageName = com.polidea.rxandroidble.sample, attributionTag = null, token = android.os.BinderProxy@5955a3e, next = null }: AdapterService getRemoteName

dariuszseweryn, 2022-06-30:
It should be just a formality but you could check `fix/sample_scan_device_name` branch — it should be working fine. I should merge it somewhere this weekend probably if it works for you.

pascalpm, 2022-07-01:
Thanks, but this did not work for me. 
Please note that this permission issue happens when you hit the 'START SCAN' button, not when selecting a Bluetooth device! No list is displayed!

dariuszseweryn, 2022-07-01:
I have changed the list adapter to use device name from the advertisement class which should not need the permission I think. Unless there is something else now. Could you share what is the exception now?

pascalpm, 2022-07-01:
When run for the first time a message will show:
'Allow sample to find, connect to and determine the relative position of nearby devices?  Allow/Don't allow'

After pressing 'Allow' the exception is generated in the debugger.

dariuszseweryn, 2022-07-01:
There should be no more access to the device name in the code of ScanActivity. Are you able to pinpoint the place / stacktrace of the exception?

I would do it myself but I don't have access to Android 12 phone

pascalpm, 2022-07-01:
Thanks for following this through.
As far as I can see the exception is created after executing 'resultAdapter.setOnAdapterItemClickListener(.....' in configureResultList(){...} (ScanPermission.java).

pascalpm, 2022-07-01:
I am happy to give you remote access to my PC if you want. If so please send me an email (pascalmater@gmail.com).

dariuszseweryn, 2022-07-02:
I think I might have found the problem. Could you pull `fix/sample_scan_device_name` branch again and re-check?
Thank you for proposing — I hope not to have to take it :)

pascalpm, 2022-07-02:
Yes, you did. Thank you very much! Should I close this issue or do you want to add some additional comment?
Again thanks for your help. I can finish my project now.

closed 2022-07-05 by dariuszseweryn