dariuszseweryn/RxAndroidBle #803: what`s wrong

JsonDing, 2022-07-07:
2022-07-07 13:07:44.705 20540-20540/com.alpha.custom E/dalvikvm: Could not find class 'android.bluetooth.le.ScanFilter$Builder', referenced from method com.polidea.rxandroidble2.internal.scan.AndroidScanObjectsConverter.toNative
2022-07-07 13:07:44.706 20540-20540/com.alpha.custom E/dalvikvm: Could not find class 'android.bluetooth.le.ScanSettings$Builder', referenced from method com.polidea.rxandroidble2.internal.scan.AndroidScanObjectsConverter.toNativeSettings
2022-07-07 13:07:44.708 20540-20540/com.alpha.custom E/dalvikvm: Could not find class 'android.bluetooth.le.ScanResult', referenced from method com.polidea.rxandroidble2.internal.scan.BackgroundScannerImpl.onScanResultReceived
2022-07-07 13:07:44.795 20540-20540/com.alpha.custom E/dalvikvm: Could not find class 'android.view.WindowInsets', referenced from method androidx.core.view.WindowInsetsCompat.toWindowInsetsCompat
2022-07-07 13:07:44.860 20540-20540/com.alpha.custom E/dalvikvm: Could not find class 'android.graphics.drawable.RippleDrawable', referenced from method androidx.appcompat.widget.AppCompatImageHelper.hasOverlappingRendering
2022-07-07 13:07:44.903 20540-20540/com.alpha.custom E/dalvikvm: Could not find class 'android.view.textclassifier.TextClassificationManager', referenced from method androidx.appcompat.widget.AppCompatTextClassifierHelper.getTextClassifier
2022-07-07 13:07:44.927 20540-20540/com.alpha.custom E/dalvikvm: Could not find class 'androidx.core.app.ActivityCompat$SharedElementCallback21Impl', referenced from method androidx.core.app.ActivityCompat.setEnterSharedElementCallback
2022-07-07 13:07:44.927 20540-20540/com.alpha.custom E/dalvikvm: Could not find class 'androidx.core.app.ActivityCompat$SharedElementCallback21Impl', referenced from method androidx.core.app.ActivityCompat.setExitSharedElementCallback
2022-07-07 13:07:45.030 20540-20540/com.alpha.custom E/: appName=com.alpha.custom, acAppName=/system/bin/surfaceflinger
2022-07-07 13:07:45.030 20540-20540/com.alpha.custom E/: 0
2022-07-07 13:08:18.695 20671-20671/com.alpha.custom E/dalvikvm: Could not find class 'android.bluetooth.le.ScanFilter$Builder', referenced from method com.polidea.rxandroidble2.internal.scan.AndroidScanObjectsConverter.toNative
2022-07-07 13:08:18.697 20671-20671/com.alpha.custom E/dalvikvm: Could not find class 'android.bluetooth.le.ScanSettings$Builder', referenced from method com.polidea.rxandroidble2.internal.scan.AndroidScanObjectsConverter.toNativeSettings
2022-07-07 13:08:18.699 20671-20671/com.alpha.custom E/dalvikvm: Could not find class 'android.bluetooth.le.ScanResult', referenced from method com.polidea.rxandroidble2.internal.scan.BackgroundScannerImpl.onScanResultReceived
2022-07-07 13:08:18.795 20671-20671/com.alpha.custom E/dalvikvm: Could not find class 'android.view.WindowInsets', referenced from method androidx.core.view.WindowInsetsCompat.toWindowInsetsCompat
2022-07-07 13:08:18.890 20671-20671/com.alpha.custom E/dalvikvm: Could not find class 'android.graphics.drawable.RippleDrawable', referenced from method androidx.appcompat.widget.AppCompatImageHelper.hasOverlappingRendering
2022-07-07 13:08:18.944 20671-20671/com.alpha.custom E/dalvikvm: Could not find class 'android.view.textclassifier.TextClassificationManager', referenced from method androidx.appcompat.widget.AppCompatTextClassifierHelper.getTextClassifier
2022-07-07 13:08:18.965 20671-20671/com.alpha.custom E/dalvikvm: Could not find class 'androidx.core.app.ActivityCompat$SharedElementCallback21Impl', referenced from method androidx.core.app.ActivityCompat.setEnterSharedElementCallback
2022-07-07 13:08:18.966 20671-20671/com.alpha.custom E/dalvikvm: Could not find class 'androidx.core.app.ActivityCompat$SharedElementCallback21Impl', referenced from method androidx.core.app.ActivityCompat.setExitSharedElementCallback
2022-07-07 13:08:19.084 20671-20671/com.alpha.custom E/: appName=com.alpha.custom, acAppName=/system/bin/surfaceflinger
2022-07-07 13:08:19.084 20671-20671/com.alpha.custom E/: 0

JsonDing, 2022-07-07:
sorry

dariuszseweryn, 2022-07-07:
I guess that you did not run the library on an Android OS hence lack of android classes. I cannot tell more. This does not seem to be a problem with the library.