dariuszseweryn/RxAndroidBle #805: Android 12 / targetSDK 31 still requires ACCESS_FINE_LOCATION, should work without it

danielRi, 2022-07-15:
**Describe the bug**
Location Permission for Android 12 is optional (targetSDK needs to be >= 31). See https://developer.android.com/guide/topics/connectivity/bluetooth/permissions
The library however forces FINE_LOCATION to be set.

**To Reproduce**
Steps to reproduce the behavior:
1. Remove `
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />` from Manifest
2. Increase targetSDK to at least 31
3. Start App without asking for location runtime permission
4. Perform any bluetooth related scan with the library. e.G. scanning
5. An error is thrown `Caused by: com.polidea.rxandroidble2.exceptions.BleScanException: Location Permission missing (code 3)`

**Expected behavior**
For Android 12 Devices, when targetting API 31, setting location is optional. The library should not force location permission to be set.

**Smartphone (please complete the following information):**
 - Device: Samsung S22 Ultra
 - OS: Android 12
 - Library version: [e.g. 1.13.1]

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
To turn on logs use:
```
`Caused by: com.polidea.rxandroidble2.exceptions.BleScanException: Location Permission missing (code 3)`
```

**Additional context**
    compileSdkVersion = 32
    buildToolsVersion = "30.0.2"
    minSdkVersion = 24
    targetSdkVersion = 32

dariuszseweryn, 2022-07-16:
Do you have `BLUETOOTH_SCAN` permission granted?

danielRi, 2022-07-18:
hey @dariuszseweryn :)

No I have not. I see this fixes the issue. Thank you!

dariuszseweryn, 2022-07-19:
Yeah, I will need to think if an alternative wording should be used or whether it would be a braking change