dariuszseweryn/RxAndroidBle #809: Establish connection trigger Pairing/Bond request every time at every session on Android 13

datsuk000, 2022-08-26:
In our BT device, the BLE pairing request is triggered when the client(app) is trying to access the "write" characteristic, but I noticed that `rxBleDevice?.establishConnection(false)` triggered it much faster and it appears every app session with Android 13. Do you know what can trigger it? And if yes, how I can avoid it at the beginning?

dariuszseweryn, 2022-08-26:
Usually bonding request is issued automatically whenever the peripheral requests bonding or the app will get an "unauthorised access" status code.
In earlier versions of Android it was possible to hijack system intents for bluetooth bonding but I am unsure if this is still possible.

datsuk000, 2022-08-26:
Thanks for the answer. Maybe you have some references where it is possible to look at how to implement this workaround?

dariuszseweryn, 2022-08-27:
Check Wiki FAQ for bonding

closed 2022-11-24 by dariuszseweryn