dariuszseweryn/RxAndroidBle #810: Throws TimeoutException after successful establishment.

neslxzhen, 2022-09-16:
**Describe the bug**
Use the `.timeout(5, TimeUnit.SECONDS)` operator to disconnect after 5 seconds of successful connection.

**To Reproduce**
Code to reproduce the behavior:
```
rxBleDevice
    .establishConnection(autoConnect = false).timeout(5, TimeUnit.SECONDS)
    .doOnError { Log.e("establishConnection", it.stackTraceToString()) }
```

**Expected behavior**
Attempt to establish a connection for 5 seconds. If the establishment is successful, no TimeoutException is thrown. If the establishment fails, a TimeoutException is thrown.

**Smartphone (please complete the following information):**
 - Device: Sony Xperia III
 - OS: Android 12
 - Library version: `com.polidea.rxandroidble2:rxandroidble:1.12.1`

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
```
establishConnection: java.util.concurrent.TimeoutException: The source did not signal an event for 5 seconds and has been terminated.
```

dariuszseweryn, 2022-09-18:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users. Also feel free to check already posted questions as a similar one may be already answered.

What you experience is expected behaviour of RxJava API operator `.timeout(int, TimeUnit)`. To achieve what you want you should use a different one or an assembly of several operators.

Best Regards

closed 2022-09-18 by dariuszseweryn

wangjianxiang, 2024-04-17:
I have encountered the same problem. How can I solve the problem of modifying DeviceModule connect timeout DEFAULT-CONNECT-TIMEOUT=35 default value