dariuszseweryn/RxAndroidBle #811: Production build crash for RxBleLog

e-Joe, 2022-09-21:
## :bomb: Steps to reproduce

1. Add custom log in RxBleClient.updateLogOptions
    `        RxBleClient.updateLogOptions(
            LogOptions.Builder()
                .setLogLevel(LogConstants.VERBOSE)
                .setMacAddressLogSetting(LogConstants.MAC_ADDRESS_FULL)
                .setUuidsLogSetting(LogConstants.UUIDS_FULL)
                .setShouldLogScannedPeripherals(true)
                .setShouldLogAttributeValues(true)
                .setLogger(MyAppBleLogger())
                .build()
        )`
2. Enable proguard in gradle build for production  It works well with debug build type .
3. When I try some BLE operation I am getting error => **Synthetic stacktrace didn't have enough elements: are you using proguard?**  Please check report from Firebase.
    

[RxBleLog crahs.txt](https://github.com/dariuszseweryn/RxAndroidBle/files/9617237/RxBleLog.crahs.txt)

dariuszseweryn, 2022-09-23:
Hello!
Thank you for using the lib and for the report. This is indeed a bug. I will work on it.

dariuszseweryn, 2022-09-25:
Could you try out 1.16.1-SNAPSHOT release and report if that fixes the issue you encountered?
Instructions how to integrate snapshot repositories are [here](https://github.com/dariuszseweryn/RxAndroidBle#snapshot).

dariuszseweryn, 2022-10-03:
@e-Joe — any chance of you testing the snapshot release?

PR #814 (2022-11-24): Fixed tag message generation with Proguard use
(Hopefully) Fixes #811
closed 2022-11-24 by dariuszseweryn