dariuszseweryn/RxAndroidBle #812: Got no connection errors when using autoconnect: true

danielRi, 2022-10-20:
This might be a misconception by me.

We noticed when we use autoconnect: true our Observable chain does not get any error callbacks when we are connecting. We have the following problem: We want to always reconnect as quickly as possible when connection is disrupted, however when the user connects the first time (with an action, like a button click), we want to show an error first and let him connect again.

```
rxBleDevice.establishConnection(autoConnect)
       .retryWhen { errorObservable ->
            return errorObservable.flatMap { error ->
               // we only get errors here with autoconnect: false, not true

               // decied to retry etc...
               return@flatMap Observable.error<RxBleConnection>(error)
            }
       }
```
Now it's already mentioned autoconnect might be confusing. From what I understand the problem is that the establishConnection never emits a connection observable, because a connection is never established.
Is there a way to handle errors that happen before a connection is created?

Note: The error occuring is not important for me here, Im only concerned about handling errors (before a connection is established)

dariuszseweryn, 2022-10-20:
What errors are you expecting?

danielRi, 2022-10-20:
GATT_ERROR 133, when I remove the battery from the device while connecting.

dariuszseweryn, 2022-10-20:
You probably won't get it when using autocorrect as the Android will continue until connection is successfully made

dariuszseweryn, 2022-10-20:
If you want to have a 30 second timeout then you should use `autoconnect=false`. 
Why do you want to use `autoconnect=true`?

closed 2022-11-07 by dariuszseweryn