dariuszseweryn/RxAndroidBle #813: Support multiple instances of the same service

spkersten, 2022-11-08:
The BLE spec allows for several instances of a service or characteristic, however the library assumes there is a unique service or characteristic with a given id.

From Core v4.2 section 3.1 on page 530:
> A device or higher level specification ... may have multiple service definitions with the same service UUID.

Android seems to support this as evident by the documentation on `BluetoothGattService#getInstanceId`:
https://developer.android.com/reference/android/bluetooth/BluetoothGattService#getInstanceId()

dariuszseweryn, 2022-11-08:
Hello, thanks for using the library. 
As far as I remember you can use all characteristics you want. If there are multiple with the same UUID then you need to explicitly call `RxBleConnection.discoverServices()`, sift through them and use `RxBleConnection.readCharacteristic(BluetoothGattCharacteristic)` API — no?

spkersten, 2022-11-08:
Yes, I missed that.

closed 2022-11-08 by spkersten