dariuszseweryn/RxAndroidBle #815: Android 12: Duplicate declarations of android.permission.ACCESS_FINE_LOCATION w different maxSdkVersions.

massivespace, 2022-12-03:
**Describe the bug**
Adding the ACCESS_FINE_LOCATION permission with a maxSdkVersion="30" into my manifest, and then attempt to upload to the play store, it fails with error "Duplicate declarations of permission android.permission.ACCESS_FINE_LOCATION with different maxSdkVersions."

**To Reproduce**
Setup manifest like this:
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" tools:targetApi="31"/>
I need it configured this way so that I can request location permissions for runtime APIs less than 31 (Android 12) and request the Bluetooth perms otherwise.  I prefer to not to have FINE location listed for SDKs greater than 23, because Android 12 also introduced a new rule that if you request FINE in API 31, you also must request COARSE.  This may only be a runtime rule, but I would just like to cleanly have the APK only request FINE permission when SDK is <= 30 and so that the "Location" permission doesn't show up under the App Permissions settings.

**Expected behavior**
Successful merge with RxAndroidBle permissions and no "Location" permission showing up on Android 12 devices under app's permissions settings.

**Smartphone (please complete the following information):**
 - Samsung S9
 - Android 12
 - Library version: 1.17.2

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
Here are the merge logs:
uses-permission-sdk-23#android.permission.ACCESS_COARSE_LOCATION
ADDED from [com.polidea.rxandroidble2:rxandroidble:1.17.2] /Users/me/.gradle/caches/transforms-2/files-2.1/        adc496cbbc6ef67b28ba9b5e84b36acc/jetified-rxandroidble-1.17.2/AndroidManifest.xml:18:5-88
    android:name
        ADDED from [com.polidea.rxandroidble2:rxandroidble:1.17.2] /Users/me/.gradle/caches/transforms-2/files-2.1/adc496cbbc6ef67b28ba9b5e84b36acc/jetified-rxandroidble-1.17.2/AndroidManifest.xml:18:29-85
uses-permission-sdk-23#android.permission.ACCESS_FINE_LOCATION
ADDED from [com.polidea.rxandroidble2:rxandroidble:1.17.2] /Users/csbyers/.gradle/caches/transforms-2/files-2.1/        adc496cbbc6ef67b28ba9b5e84b36acc/jetified-rxandroidble-1.17.2/AndroidManifest.xml:19:5-86
    android:name
        ADDED from [com.polidea.rxandroidble2:rxandroidble:1.17.2]

**Additional context**
While I believe I can just remove the FINE location (and probably the other permissions as well) from my manifest, it seems like you would want to change:
<uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION"/>
to:
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="30"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30"/>
and maybe update the library to min sdk of 23, but I'm not sure.  Would it maybe be possible to combine uses-permission-sdk-23 with maxSdkVersion?
These changes to Bluetooth Google keeps making get messier and harder to maintain.  Let's hope these are the last permissions changes they make for a while.  :(

dariuszseweryn, 2022-12-03:
Hello,
Have you read [Permissions](https://github.com/dariuszseweryn/RxAndroidBle#permissions) part of the Readme? AFAIK it is described there what to do in case of permissions mismatch.

massivespace, 2022-12-03:
I read it right down to "Additionally, in a future version of RxAndroidBle, these permissions will be restricted to only APIs 23-30." so I assumed it was not yet implemented and skipped further reading.  I apologize, so I see the text below that addresses my need.  Thank you very much!

closed 2022-12-03 by massivespace