dariuszseweryn/RxAndroidBle #816: Refresh list Characteristic

ThuanNH4, 2022-12-13:
hi Darius, 
I have 1 question
When i make 1 request to change mode to my device, my device already changed mode (new characteristic has been added) but then i make 1 request to read data from that new characteristic, I got the exception com.polidea.rxandroidble2.exceptions.Ble CharacteristicNotFoundException: No trait found with UUID xxx
I think I will need send a refresh request to my device before reading, can you guide me how to refresh the current list Characteristic?

dariuszseweryn, 2022-12-14:
[Would this help](https://github.com/dariuszseweryn/RxAndroidBle/pull/453#issuecomment-459787554)?

ThuanNH4, 2022-12-14:
Exactly, let me check. 
Thank you

closed 2022-12-14 by ThuanNH4