dariuszseweryn/RxAndroidBle #818: ERROR: com.polidea.rxandroidble.internal.operations.ScanOperation.protectedRun

utsavDave97, 2023-01-17:
Getting the following crash for Android 12 and above when trying to scan for Bluetooth devices in Flutter.

# com.polidea.rxandroidble.internal.operations.ScanOperation.protectedRun

```
Exception java.lang.IllegalStateException: Fatal Exception thrown on Scheduler.Worker thread.
  at rx.internal.schedulers.ScheduledAction.run (ScheduledAction.java)
  at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run (ExecutorScheduler.java)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:920)
Caused by k.n.e: Error occurred when trying to propagate error to Observer.onError
  at rx.observers.SafeSubscriber._onError (SafeSubscriber.java)
  at rx.observers.SafeSubscriber.onError (SafeSubscriber.java)
  at rx.internal.operators.OperatorUnsubscribeOn$1.onError (OperatorUnsubscribeOn.java)
  at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.onError (OperatorSubscribeOn.java)
  at rx.internal.operators.OnSubscribeCreate$BaseEmitter.onError (OnSubscribeCreate.java)
  at com.polidea.rxandroidble.internal.operations.ScanOperation.protectedRun (ScanOperation.java)
  at com.polidea.rxandroidble.internal.QueueOperation$1.call (QueueOperation.java)
  at com.polidea.rxandroidble.internal.QueueOperation$1.call (QueueOperation.java)
  at rx.internal.operators.OnSubscribeCreate.call (OnSubscribeCreate.java)
  at rx.internal.operators.OnSubscribeCreate.call (OnSubscribeCreate.java)
  at rx.Observable.unsafeSubscribe (Observable.java)
  at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.call (OperatorSubscribeOn.java)
  at rx.internal.schedulers.ScheduledAction.run (ScheduledAction.java)
Caused by k.n.a: 2 exceptions occurred. 
  at rx.observers.SafeSubscriber._onError (SafeSubscriber.java)
Caused by k.n.a$a: Chain of Causes for CompositeException In Order Received =>
  at com.google.firebase.crashlytics.internal.stacktrace.TrimmedThrowableData.<init> (TrimmedThrowableData.java)
  at com.google.firebase.crashlytics.internal.stacktrace.TrimmedThrowableData.<init> (TrimmedThrowableData.java)
  at com.google.firebase.crashlytics.internal.stacktrace.TrimmedThrowableData.<init> (TrimmedThrowableData.java)
  at com.google.firebase.crashlytics.internal.common.CrashlyticsReportDataCapture.captureEventData (CrashlyticsReportDataCapture.java)
  at com.google.firebase.crashlytics.internal.common.SessionReportingCoordinator.persistEvent (SessionReportingCoordinator.java)
  at com.google.firebase.crashlytics.internal.common.SessionReportingCoordinator.persistFatalEvent (SessionReportingCoordinator.java)
  at com.google.firebase.crashlytics.internal.common.CrashlyticsController$2.call (CrashlyticsController.java)
  at com.google.firebase.crashlytics.internal.common.CrashlyticsController$2.call (CrashlyticsController.java)
  at com.google.firebase.crashlytics.internal.common.CrashlyticsBackgroundWorker$3.then (CrashlyticsBackgroundWorker.java)
  at com.google.android.gms.tasks.zzf.run (zzf.java)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at com.google.firebase.crashlytics.internal.common.ExecutorUtils$1$1.onRun (ExecutorUtils.java)
  at com.google.firebase.crashlytics.internal.common.BackgroundPriorityRunnable.run (BackgroundPriorityRunnable.java)
Caused by c.f.c.k0.n:
  at com.polidea.rxandroidble.internal.operations.ScanOperation.protectedRun (ScanOperation.java)
  at com.polidea.rxandroidble.internal.QueueOperation$1.call (QueueOperation.java)
  at com.polidea.rxandroidble.internal.QueueOperation$1.call (QueueOperation.java)
  at rx.internal.operators.OnSubscribeCreate.call (OnSubscribeCreate.java)
  at rx.internal.operators.OnSubscribeCreate.call (OnSubscribeCreate.java)
  at rx.Observable.unsafeSubscribe (Observable.java)
  at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.call (OperatorSubscribeOn.java)
  at rx.internal.schedulers.ScheduledAction.run (ScheduledAction.java)
  at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run (ExecutorScheduler.java)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
Caused by java.lang.RuntimeException:
  at io.flutter.embedding.engine.FlutterJNI.ensureRunningOnMainThread (FlutterJNI.java)
  at io.flutter.embedding.engine.FlutterJNI.dispatchPlatformMessage (FlutterJNI.java)
  at io.flutter.embedding.engine.dart.DartMessenger.send (DartMessenger.java)
  at io.flutter.embedding.engine.dart.DartExecutor$DefaultBinaryMessenger.send (DartExecutor.java)
  at io.flutter.embedding.engine.dart.DartExecutor.send (DartExecutor.java)
  at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler$EventSinkImplementation.error (EventChannel.java)
  at com.polidea.flutter_ble_lib.event.ScanningStreamHandler.onError (ScanningStreamHandler.java)
  at com.polidea.flutter_ble_lib.FlutterBleLibPlugin$4.onError (FlutterBleLibPlugin.java)
  at com.polidea.multiplatformbleadapter.BleModule$19.call (BleModule.java)
  at com.polidea.multiplatformbleadapter.BleModule$19.call (BleModule.java)
  at rx.internal.util.ActionSubscriber.onError (ActionSubscriber.java)
  at rx.observers.SafeSubscriber._onError (SafeSubscriber.java)
  at rx.observers.SafeSubscriber.onError (SafeSubscriber.java)
  at rx.observers.Subscribers$5.onError (Subscribers.java)
  at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError (OperatorMerge.java)
  at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate (OperatorMerge.java)
  at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop (OperatorMerge.java)
  at rx.internal.operators.OperatorMerge$MergeSubscriber.emit (OperatorMerge.java)
  at rx.internal.operators.OperatorMerge$InnerSubscriber.onError (OperatorMerge.java)
  at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError (OnSubscribeMap.java)
  at rx.internal.operators.OperatorUnsubscribeOn$1.onError (OperatorUnsubscribeOn.java)
  at rx.internal.operators.OnSubscribeCreate$BaseEmitter.onError (OnSubscribeCreate.java)
  at rx.internal.util.ObserverSubscriber.onError (ObserverSubscriber.java)
  at rx.observers.SafeSubscriber._onError (SafeSubscriber.java)
  at rx.observers.SafeSubscriber.onError (SafeSubscriber.java)
  at rx.internal.operators.OperatorUnsubscribeOn$1.onError (OperatorUnsubscribeOn.java)
  at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.onError (OperatorSubscribeOn.java)
  at rx.internal.operators.OnSubscribeCreate$BaseEmitter.onError (OnSubscribeCreate.java)
  at com.polidea.rxandroidble.internal.operations.ScanOperation.protectedRun (ScanOperation.java)
```

dariuszseweryn, 2023-01-17:
Do you have a reproduce scenario? I'm not very familiar with the Flutter library nor I see any clear exception here.

closed 2023-01-18 by utsavDave97

dariuszseweryn, 2023-01-18:
@utsavDave97 could you add more information on why you have closed this issue? That could benefit other users in the future.

utsavDave97, 2023-01-18:
@dariuszseweryn Hey sorry for that. To be honest even I don't have any idea to reproduce this. But it is mainly happening for Android 12 and above. For my project I just decided to go with different library as I was not able to find a solution for this error.