dariuszseweryn/RxAndroidBle #819: Unable to send BLE commands to device

nmoges, 2023-02-07:
Hello,

I'm facing some issue for which I didn't find a solution yet. On my project I'm supposed to connect to a device through BLE connection and read/write data including from CBOR files.
I don't have any trouble to do so, but the issue is that all my BLE commands seem to don't work anymore after I read a CBOR file.
In the log below, I tried two operations after reading a CBOR file (which is normally correctly closed) :
- setup a characteristic notification (00010012-cad9-46c6-a2ea-2ca16d57b4a5) -> Got an BleCannotSetCharacteristicNotificationException after few seconds
- write a data throught a characteristic (00010011-cad9-46c6-a2ea-2ca16d57b4a5) -> Got a status 4 (GATT_INVALID_PDU)

```
Setup notification : 00010012-cad9-46c6-a2ea-2ca16d57b4a5
Write on 00010011-cad9-46c6-a2ea-2ca16d57b4a5 | Data : 01 MAC ADDRESS : E8:31:CD:7B:3A:59
Device connection state : RxBleConnectionState{CONNECTED} | Connection is null : false
setCharacteristicNotification() - uuid: 00010012-cad9-46c6-a2ea-2ca16d57b4a5 enable: false
com.polidea.rxandroidble2.exceptions.BleCannotSetCharacteristicNotificationException: Cannot write client characteristic config descriptor (code 3) with characteristic UUID 00010012-cad9-46c6-a2ea-2ca16d57b4a5
  	at com.polidea.rxandroidble2.internal.connection.NotificationAndIndicationManager.lambda$writeClientCharacteristicConfig$9(NotificationAndIndicationManager.java:194)
  	at com.polidea.rxandroidble2.internal.connection.NotificationAndIndicationManager$$ExternalSyntheticLambda4.apply(Unknown Source:4)
  	at io.reactivex.internal.operators.completable.CompletableResumeNext$ResumeNextObserver.onError(CompletableResumeNext.java:82)
 	at io.reactivex.internal.operators.observable.ObservableIgnoreElementsCompletable$IgnoreObservable.onError(ObservableIgnoreElementsCompletable.java:61)
  	at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
  	at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1$1.onError(FIFORunnableEntry.java:73)
  	at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:70)
  	at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
 	at com.polidea.rxandroidble2.internal.util.QueueReleasingEmitterWrapper.onError(QueueReleasingEmitterWrapper.java:47)
 	at io.reactivex.internal.observers.DeferredScalarDisposable.error(DeferredScalarDisposable.java:100)
  	at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onError(SingleToObservable.java:77)
  	at io.reactivex.internal.disposables.EmptyDisposable.error(EmptyDisposable.java:78)
  Caused by: com.polidea.rxandroidble2.exceptions.BleGattCallbackTimeoutException: GATT exception from MAC address E8:31:CD:7B:3A:59, with type BleGattOperation{description='DESCRIPTOR_WRITE'}
  	at com.polidea.rxandroidble2.internal.SingleResponseOperation.timeoutFallbackProcedure(SingleResponseOperation.java:92)
  	at com.polidea.rxandroidble2.internal.SingleResponseOperation.protectedRun(SingleResponseOperation.java:54)
  	at com.polidea.rxandroidble2.internal.QueueOperation$1.subscribe(QueueOperation.java:41)
  	at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
  	at io.reactivex.Observable.subscribe(Observable.java:12284)
 	at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn.subscribeActual(ObservableUnsubscribeOn.java:32)
  	at io.reactivex.Observable.subscribe(Observable.java:12284)
  	at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1.run(FIFORunnableEntry.java:56)
  Error writing 00010011-cad9-46c6-a2ea-2ca16d57b4a5
  Error writing GATT exception from MAC='E8:31:CD:7B:3A:59', status 4 (GATT_INVALID_PDU), type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x04 here https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/include/gatt_api.h)
```
Any idea ?

Thanks in advance :)

 - Device: Xiaomi Mi 9T
 - OS: Android 10
 - Library version: 1.17.0

dariuszseweryn, 2023-02-07:
Can you setup the notification before reading the CBOR file?

nmoges, 2023-02-07:
Thanks for the reply. Unfortunately in my scenario it's not possible. The steps are that I'm supposed to read a specific CBOR file first while displaying to user a loading screen, and then setup the notification on this characteristic.

dariuszseweryn, 2023-02-08:
Can you confirm that this flow works using a different setup? Different phone vendor/OS/app?

Given the maturity of the BLE I am inclined to think that either the peripheral has an implementation bug or there is a missing step in the usage procedure i.e. the peripheral expects something more to happen before the notification can be setup.

nmoges, 2023-02-08:
T tried with a Huawei P20 on Android 8 too, same issue. I might have miss a step indeed. What's strange is that anything that's not CBOR-related doesn't work after reading a CBOR file. Which means that I can't setup a notification or send any BLE command, except to read or write a CBOR files.
I'll try to investigate deeper on my side.

nmoges, 2023-03-04:
Sorry for the delay. Unfortunately no, I had to focus on other features to deliver for my deadline, I'm planning to work deeply on this issue on the next few days. But after a first investigation with others dev on my team, the problem is more related to the BLE module I implemented in my app and how I use it (we don't face the same problem in others projects in which the same module is implemented)

nmoges, 2023-08-21:
Hello @dariuszseweryn, 
Sorry for the late update. Yes we solved the issue on our side. The problem was the method we used to read the CBOR file. Some resources were not correctly disposed after reading, making the BLE connection always "busy", that's why we weren't able to send any BLE commands after that. 
We can close this issue.

dariuszseweryn, 2023-08-21:
Thank you for the info! Closing

closed 2023-08-21 by dariuszseweryn