dariuszseweryn/RxAndroidBle #821: Bluetooth beacon not found

junwu-lyft, 2023-02-28:
**Describe the bug**
I purchased a BlueCharm BC011 beacon(https://bluecharmbeacons.com/product/bluetooth-ble-multi-beacon-bc011/)  from Amazon. I'm trying to make a demo based on it. But no matter which mode I config it as (iBeacon, Eddystone-UID, Eddystone-URL), I can't find it in the scanning results all the time.

- There are many bluetooth devices found in the scanning, but I checked the MAC address, non of them are from the one I purchased.
- I'm sure the beacon works  because 3rd party apps can find it.

Any thought/hints on it? Thanks!

**Smartphone (please complete the following information):**
Oneplus 7T Pro
Android 13

dariuszseweryn, 2023-02-28:
How do you know that it is not showing up in the scan results?

junwu-lyft, 2023-02-28:
I print all the MAC addresses of scanned devices using a for loop. Couldn't find the one I purchased.
The MAC address of the purchased beacon is written on the tag, it's also correctly displayed by other apps.

dariuszseweryn, 2023-02-28:
How do you run the scan?

junwu-lyft, 2023-02-28:
```
import com.polidea.rxandroidble2.RxBleClient
...

 private val rxBleClient: RxBleClient

...
                rxBleClient.scanBleDevices(
                    scanSettings
                ).map { scanResult ->
                    // print the scan Result
               }
```

scanSettings:
- matchMode = 1
- numOfMatches = 3
-  SanMode = tested low latency, balanced, low power, none of them work

Besides that I found that all of `uuid` in scanResult is empty. I got this error when converting it to a string:
`uuid convert error java.lang.NullPointerException: rxBleDevice.bluetoothDevice.uuids must not be null`

Is this normal?

junwu-lyft, 2023-03-01:
I/DEBUG_BLE: MacAddress = CC:D2:81:70:D6:26 name = null
 I/DEBUG_BLE: bluetoothDevices = CC:D2:81:70:D6:26
 I/DEBUG_BLE: uuid convert error java.lang.NullPointerException: rxBleDevice.bluetoothDevice.uuids must not be null
 I/DEBUG_BLE: MacAddress = 08:E3:4D:91:29:76 name = null
 I/DEBUG_BLE: bluetoothDevices = 08:E3:4D:91:29:76
 I/DEBUG_BLE: uuid convert error java.lang.NullPointerException: rxBleDevice.bluetoothDevice.uuids must not be null
 I/DEBUG_BLE: MacAddress = 71:84:6F:C3:CA:DE name = null
 I/DEBUG_BLE: bluetoothDevices = 71:84:6F:C3:CA:DE
 I/DEBUG_BLE: uuid convert error java.lang.NullPointerException: rxBleDevice.bluetoothDevice.uuids must not be null
 I/DEBUG_BLE: MacAddress = 3F:E2:F9:19:44:D4 name = null
 I/DEBUG_BLE: bluetoothDevices = 3F:E2:F9:19:44:D4
 I/DEBUG_BLE: uuid convert error java.lang.NullPointerException: rxBleDevice.bluetoothDevice.uuids must not be null
 I/DEBUG_BLE: MacAddress = 9C:58:3C:D8:9C:CF name = null
 I/DEBUG_BLE: bluetoothDevices = 9C:58:3C:D8:9C:CF
 I/DEBUG_BLE: uuid convert error java.lang.NullPointerException: rxBleDevice.bluetoothDevice.uuids must not be null
 I/DEBUG_BLE: MacAddress = F6:FB:C9:E1:F5:7F name = Hue color lamp
 I/DEBUG_BLE: bluetoothDevices = F6:FB:C9:E1:F5:7F

[image]

Both Eddystone beacon&iBeacon can't be found.

dariuszseweryn, 2023-03-01:
> Besides that I found that all of uuid in scanResult is empty. I got this error when converting it to a string:
uuid convert error java.lang.NullPointerException: rxBleDevice.bluetoothDevice.uuids must not be null
> 
> Is this normal?

UUIDs may be null

Could you show only logs from the library itself with DEBUG log level? From the beginning of the scan.

dariuszseweryn, 2023-03-02:
Hmmm... I found [this information](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#assert-never-for-location) which is why you may not see your beacon in the scan results:
> Note: If you include neverForLocation in your android:usesPermissionFlags, some BLE beacons are filtered from the scan results.

junwu-lyft, 2023-03-07:
Hi Dariusz,
I root caused the issue, the answer is in README:

If you want to derive the user's location from BLE scanning in API >= 31
API 31 (Android 12) introduced new Bluetooth permissions. RxAndroidBle uses the android:usesPermissionFlags="neverForLocation" attribute on the BLUETOOTH_SCAN permission, which indicates scanning will not be used to derive the user's location, so location permissions are not required. If you need to locate the user with BLE scanning, use this instead, but keep in mind that you will still need ACCESS_FINE_LOCATION:
`
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:node="replace" />`

Thanks for your support! I'll close this issue.

closed 2023-03-07 by junwu-lyft