dariuszseweryn/RxAndroidBle #822: Permission for connection with sample-kotlin

SeigiTenshi, 2023-03-03:
Hi,

I use yours sample-kotlin to try understand how to code with your library but I face an issues.
The scan work perfectly fine but when I try "service discovery" I have this message : "Connection error: java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNEC..."
I had <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/> in AndroidManifest.xml but the issue is the same.

My mobile is an OppoReno 6 5g with 12 android version

Thanks for any help you can provide

dariuszseweryn, 2023-03-03:
Hmmm... I see that there is a problem in the sample app and the library. You should be able to workaround it by giving a `BLUETOOTH_CONNECT` runtime permission to the app.

SeigiTenshi, 2023-03-06:
Thanks I was able to fix it with the runtime permission it's work perfectly fine now

closed 2023-03-06 by SeigiTenshi