dariuszseweryn/RxAndroidBle #829: BleGattCharacteristicException while readCharacteristic

mkiisoft, 2023-05-14:
**Describe the bug**
I have a BLE device with several (11) characteristics and almost of those working. One in particular is not, but I was able to read it using HTML5 Bluetooth library on Android itself (same phone) and on iOS connecting to the same device (the result on iOS has several trailing zeros).

The following exception is being thrown:

com.polidea.rxandroidble2.exceptions.BleGattCharacteristicException: GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 133 (GATT_ERROR), type BleGattOperation{description='CHARACTERISTIC_READ'}. (Look up status 0x85 here https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/include/gatt_api.h)

Every single other Characteristic works, no problem, still connected (so, it's not a connection issue) right before and right after that one, except for every time I'm reading that Characteristic.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to device
2. Read Characteristic

**Expected behavior**
iOS and HTML5/Js native implementations, all of them return a List of int as expected being the following one:

`
[118, 49, 46, 56, 46, 50, 32, 104, 119, 49, 32, 110, 54, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0]
`

**Smartphone / tablet**
 - Device: OnePlus 6T
 - OS: Android 11
 - Package version: 1.17.2

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
To turn on logs use:
```
RxBleClient.updateLogOptions(new LogOptions.Builder()
        .setLogLevel(LogConstants.DEBUG)
        .setMacAddressLogSetting(LogConstants.MAC_ADDRESS_FULL)
        .setUuidsLogSetting(LogConstants.UUIDS_FULL)
        .setShouldLogAttributeValues(true)
        .build()
);
```
```
com.polidea.rxandroidble2.exceptions.BleGattCharacteristicException: GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 133 (GATT_ERROR), type BleGattOperation{description='CHARACTERISTIC_READ'}. (Look up status 0x85 here https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/include/gatt_api.h)
```

**Additional context**
I discarded being my device as I'm getting the data if I use a HTML5/Js library and the data gets as expected.

dariuszseweryn, 2023-05-14:
What is your device?
Does your device pair/bond? 
Does it changes services?
Could you share a full log from the beginning of connection up to the error?

mkiisoft, 2023-05-14:
@dariuszseweryn Hello, thanks for asking!

I have all in my bug:

Smartphone / tablet

Device: OnePlus 6T
OS: Android 11
Package version: 1.17.2

I sync to a N64 Bluetooth dongle successfully, no error, I can write and read every other characteristic except for that one with this library. It does changes services.

```
I/app (22867): EVENT: ConnectionStateUpdate(deviceId: 90:38:0C:F5:56:0A, connectionState: DeviceConnectionState.connecting, failure: null)
D/BluetoothGatt(22867): onClientConnectionState() - status=0 clientIf=10 device=90:38:0C:F5:56:0A
I/app (22867): EVENT: ConnectionStateUpdate(deviceId: 90:38:0C:F5:56:0A, connectionState: DeviceConnectionState.connected, failure: null)
I/app (22867): CONNECTED
D/BluetoothGatt(22867): requestConnectionPriority() - params: 1
D/BluetoothGatt(22867): onConnectionUpdated() - Device=90:38:0C:F5:56:0A interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(22867): onConnectionUpdated() - Device=90:38:0C:F5:56:0A interval=12 latency=0 timeout=500 status=0
D/BluetoothGatt(22867): discoverServices() - device: 90:38:0C:F5:56:0A
D/BluetoothGatt(22867): onSearchComplete() = Device=90:38:0C:F5:56:0A Status=0
I/System.out(22867): CHARACTERISTIC: 56830f56-5180-fab0-314b-2fa176799a0c
I/System.out(22867): CHARACTERISTIC_DEVICE_ID: 90:38:0C:F5:56:0A
I/System.out(22867): CHARACTERISTIC_RESULT: CharOperationSuccessful(deviceId=90:38:0C:F5:56:0A, value=[10, 86, -11, 12, 56, -112])
I/app (22867): UPDATE: CharacteristicValue(characteristic: Characteristic(characteristicId: 56830f56-5180-fab0-314b-2fa176799a0c, serviceId: 56830f56-5180-fab0-314b-2fa176799a00, deviceId: 90:38:0C:F5:56:0A), value: ([10, 86, 245, 12, 56, 144]))
I/System.out(22867): CHARACTERISTIC: 56830f56-5180-fab0-314b-2fa176799a09
I/System.out(22867): CHARACTERISTIC_MESSAGE: 90:38:0C:F5:56:0A
I/System.out(22867): ERROR: com.polidea.rxandroidble2.exceptions.BleGattCharacteristicException: GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 133 (GATT_ERROR), type BleGattOperation{description='CHARACTERISTIC_READ'}. (Look up status 0x85 here https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/include/gatt_api.h)
I/app (22867): EVENT: CharacteristicValue(characteristic: Characteristic(characteristicId: 56830f56-5180-fab0-314b-2fa176799a09, serviceId: 56830f56-5180-fab0-314b-2fa176799a00, deviceId: 90:38:0C:F5:56:0A), value: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 133 (GATT_ERROR), type BleGattOperation{description='CHARACTERISTIC_READ'}. (Look up status 0x85 here https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/include/gatt_api.h)")))
I/app (22867): UPDATE: CharacteristicValue(characteristic: Characteristic(characteristicId: 56830f56-5180-fab0-314b-2fa176799a09, serviceId: 56830f56-5180-fab0-314b-2fa176799a00, deviceId: 90:38:0C:F5:56:0A), value: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "GATT exception from MAC='XX:XX:XX:XX:XX:XX', status 133 (GATT_ERROR), type BleGattOperation{description='CHARACTERISTIC_READ'}. (Look up status 0x85 here https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/include/gatt_api.h)")))
```

dariuszseweryn, 2023-05-14:
Is [this](https://www.nintendolife.com/news/2023/01/review-blueretro-n64-adapter-this-plug-and-play-bluetooth-dongle-is-the-perfect-match-for-your-nso-pad) the peripheral you connect your phone to?

I asked for logs, please share them but with below settings in your app — it will give more information on the internals of the library. Also timings may be important so please leave timestamps.
```
RxBleClient.updateLogOptions(new LogOptions.Builder()
        .setLogLevel(LogConstants.VERBOSE)
        .setMacAddressLogSetting(LogConstants.MAC_ADDRESS_FULL)
        .setUuidsLogSetting(LogConstants.UUIDS_FULL)
        .setShouldLogAttributeValues(true)
        .build()
);
```

mkiisoft, 2023-05-14:
@dariuszseweryn Correct! I'm able to do all the operations on web with the same Android device to that dongle and with iOS. So far, only that characteristic is the one getting me an exception with this library.

Full log of that event:

```
2023-05-14 18:22:14.402 10939-31169/com.ble.app D/RxBle#ConnectionOperationQueue: FINISHED ConnectionPriorityChangeOperation(132136986) in 2012 ms
2023-05-14 18:22:14.413 972-21802/? D/vendor.qti.bluetooth@1.0-wake_lock: Release wakelock is released 
2023-05-14 18:22:14.442 10939-10939/com.ble.app D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(90065876)
2023-05-14 18:22:14.443 10939-31169/com.ble.app D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(90065876)
2023-05-14 18:22:14.444 10939-31169/com.ble.app I/RxBle#ConnectionOperationQueue: RUNNING  ServiceDiscoveryOperation{MAC='90:38:0C:F5:56:0A'}
2023-05-14 18:22:14.446 10939-31154/com.ble.app D/BluetoothGatt: discoverServices() - device: 90:38:0C:F5:56:0A
2023-05-14 18:22:14.446 21694-21711/? D/BtGatt.GattService: discoverServices() - address=90:38:0C:F5:56:0A, connId=268
2023-05-14 18:22:14.447 21694-21923/? V/bt_stack: [VERBOSE1:bta_gattc_main.cc(361)] bta_gattc_hdl_event: Event:BTA_GATTC_API_SEARCH_EVT
2023-05-14 18:22:14.447 21694-21923/? V/bt_stack: [VERBOSE1:bta_gattc_main.cc(305)] bta_gattc_sm_execute: State 0x02 [GATTC_CONN_ST], Event 0x1f09[BTA_GATTC_API_SEARCH_EVT]
2023-05-14 18:22:14.447 21694-21923/? V/bt_stack: [VERBOSE1:bta_gattc_act.cc(1030)] bta_gattc_search: conn_id=0x010c
2023-05-14 18:22:14.447 21694-21923/? V/bt_stack: [VERBOSE1:bta_gattc_main.cc(341)] bta_gattc_sm_execute: GATTC State Change: 0x02 -> 0x02 after Event 0x1f09
2023-05-14 18:22:14.447 21694-21738/? D/BtGatt.GattService: onSearchCompleted() - connId=268, status=0
2023-05-14 18:22:14.448 21694-21738/? D/bt_bta_gattc: bta_gattc_get_gatt_db
2023-05-14 18:22:14.448 21694-21738/? V/bt_stack: [VERBOSE1:bta_gattc_cache.cc(618)] bta_gattc_get_gatt_db_impl: start_handle 0x0000, end_handle 0xffff
2023-05-14 18:22:14.448 21694-21738/? D/BtGatt.GattService: onGetGattDb() - address=90:38:0C:F5:56:0A
2023-05-14 18:22:14.448 21694-21738/? D/BtGatt.GattService: got service with UUID=00001801-0000-1000-8000-00805f9b34fb id: 1
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=00002a05-0000-1000-8000-00805f9b34fb id: 3
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got service with UUID=00001800-0000-1000-8000-00805f9b34fb id: 20
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=00002a00-0000-1000-8000-00805f9b34fb id: 22
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=00002a01-0000-1000-8000-00805f9b34fb id: 24
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got service with UUID=56830f56-5180-fab0-314b-2fa176799a00 id: 64
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a01 id: 66
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a02 id: 68
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a03 id: 70
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a04 id: 72
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a05 id: 74
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a06 id: 76
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a07 id: 78
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a08 id: 80
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a09 id: 82
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a0a id: 84
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a0b id: 86
2023-05-14 18:22:14.449 21694-21738/? D/BtGatt.GattService: got characteristic with UUID=56830f56-5180-fab0-314b-2fa176799a0c id: 88
2023-05-14 18:22:14.452 10939-31013/com.ble.app D/BluetoothGatt: onSearchComplete() = Device=90:38:0C:F5:56:0A Status=0
2023-05-14 18:22:14.453 10939-31013/com.ble.app I/RxBle#GattCallback: MAC='90:38:0C:F5:56:0A'     onServicesDiscovered(), status=0
2023-05-14 18:22:14.455 10939-31170/com.ble.app V/RxBle#ServiceDiscoveryOperation: Preparing services description
2023-05-14 18:22:14.463 10939-31170/com.ble.app V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
    MAC='90:38:0C:F5:56:0A'
    PERIPHERAL NAME: BlueRetro_N64_560A
    -------------------------------------------------------------------------
    
    Primary Service - Generic Attribute (00001801-0000-1000-8000-00805f9b34fb)
    Instance ID: 1
    -> Characteristics:
    	* Service Changed (00002a05-0000-1000-8000-00805f9b34fb)
    	  Properties: [ INDICATE ]
    
    Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
    Instance ID: 20
[... cut ...]  deviceId: 90:38:0C:F5:56:0A), value: Result<List<int>, GenericFailure<CharacteristicValueUpdateError>?>.failure(GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "GATT exception from MAC='90:38:0C:F5:56:0A', status 133 (GATT_ERROR), type BleGattOperation{description='CHARACTERISTIC_READ'}. (Look up status 0x85 here https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/include/gatt_api.h)")))

```

dariuszseweryn, 2023-05-16:
If the peripheral has been paired/bonded it may be that something got off with the Android cache. Could you try reboot your phone and retry?
If that does not work, you may try to add an artificial delay between connecting and before the first operation on `RxBleConnection` happens as Android needs some time to re-establish the bond.

mkiisoft, 2023-05-16:
@dariuszseweryn I tested both scenarios and same issue with the same characteristic and on Web with the same device I can always get a result. I tried rebooting both devices several times, I also added from 2 to 10 seconds delays with several retries (10 retries) and same error. I did the same right after and right before that characteristic with any other characteristic and I have no issues, I can read and write, no problem.

dariuszseweryn, 2023-05-17:
Could you try `nRF Connect` app and see if you could perform the same scenario there?

mkiisoft, 2023-05-17:
@dariuszseweryn I get the same error "Error 133 (0x85) GATT ERROR". The crazy thing is that I just tested with the Web interface using the same Android and it works. I also tested on iOS and it works:

This is Web Android (same device):
[image]

This is the error with nRF Connect (same device):
[image]

This is iOS: 
[image]

dariuszseweryn, 2023-05-17:
It looks like an Android BLE stack bug. Not sure what the web API uses. We could try to dig deeper by getting your phone's BT HCI logs with both app usage and web usage and analysing them — maybe some workaround will pop-up

mkiisoft, 2023-05-17:
@dariuszseweryn I just tested on a Xiaomi MI9T Android 11 and same issue. Will try to dig deeper and I'll talk with the creator of the dongle to see if a firmware update can make it readable for Android.

qq1023308356, 2023-06-29:
> 挖掘，我将与加密狗的创建者交谈，看看固件更新是否可以使其在 Android

Excuse me, is there any progress on this issue?

AliAzaz, 2023-08-07:
I am also rarely getting the same sort of error:

Caused by com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)
       at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:80)
       at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:299)
       at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:864)
       at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback()
       at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:294)
       at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:128)
       at android.os.Binder.execTransactInternal(Binder.java:1285)
       at android.os.Binder.execTransact(Binder.java:1244)