dariuszseweryn/RxAndroidBle #832: GenericFailure<CharacteristicValueUpdateError>

qq1023308356, 2023-06-27:
Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "Cannot write client characteristic config descriptor (code 3) with characteristic UUID 6e400003-b5a3-f393-e0a9-e50e24dcca9e")

qq1023308356, 2023-06-27:
After the appearance, all reading and writing services are invalid, and it is useless to restart app. Need to restart the phone to be normal

dariuszseweryn, 2023-06-29:
Hello
Unfortunately your bug report does not provide much actionable information — could you conform to the template?

qq1023308356, 2023-08-17:
This problem is because the device switched UUID, and then when reading and writing, it took the previous UUID,clearGattCache this method is useless, bluetoothGatt there is no refresh method in it. set it to turn Bluetooth back on and it will return to normal. do you have any good solutions here?

qq1023308356, 2023-08-17:
https://github.com/dariuszseweryn/RxAndroidBle/issues/692
Should be similar to this question

qq1023308356, 2023-08-30:
@dariuszseweryn

dariuszseweryn, 2023-08-30:
Hello, there is a wiki page about custom operations [one of which allows for gatt refresh](https://github.com/dariuszseweryn/RxAndroidBle/wiki/Tutorial:-RxBleCustomOperation#call-vanilla-api-refresh-method-by-reflection). You can then use the next operation to discover services once more an use them accordingly. I did not had an opportunity to work with devices that change services so you need to figure out the details

qq1023308356, 2023-08-31:
@dariuszseweryn 
BluetoothGatt.refresh() This method is invalid.
Discover services bypassing library cache，How to use？

dariuszseweryn, 2023-08-31:
> BluetoothGatt.refresh() This method is invalid.

Calls using reflection don't work anymore?

> Discover services bypassing library cache，How to use？

Custom operation on the client

qq1023308356, 2023-09-01:
> > BluetoothGatt.refresh() This method is invalid.
> 
> Calls using reflection don't work anymore?
> 
> > Discover services bypassing library cache，How to use？
> 
> Custom operation on the client

Yes, the error report did not find this method.

dariuszseweryn, 2023-09-01:
Could you share logs?

closed 2024-01-07 by dariuszseweryn