dariuszseweryn/RxAndroidBle #838: Creating a Fork

MaxWebfactor, 2023-09-26:
Hi there,
first I have to say thank you for your work by creating this library.

I created a fork of your repo to test a change and then upload it via jitpack to easily integrate the fork into my project. So currently I got stuck while uploading to jitpack.

Here is the error:
* What went wrong:
Execution failed for task ':dagger-library-shadow:signMavenPublication'.
> Cannot perform signing task ':dagger-library-shadow:signMavenPublication' because it has no configured signatory

So I thought it would be easy to fork the repo, connect my fork with jitpack and then use the new created lib. But, sadly it don't work out of the box. 

Has anyone a suggestion how it will run? Which things have to change? 

I am a bit frustrated so I hope for help.

Regards Max

dariuszseweryn, 2023-09-26:
Hmmm... I am not a master of grade myself. You may first of all remove signing as it is not needed for jitpack as far as I can tell.
As for using the project as a module you could look how the `sample` app is using `rxandroidble`:
```
dependencies {
    implementation project(path: ':rxandroidble')
    implementation rootProject.ext.libs.appcompat
    implementation rootProject.ext.libs.material_design
    implementation 'com.jakewharton:butterknife:10.2.3'
    annotationProcessor 'com.jakewharton:butterknife-compiler:10.2.3'
    implementation rootProject.ext.libs.rxandroid
    implementation rootProject.ext.libs.rxjava2
    implementation rootProject.ext.libs.rxjava_binding
    implementation rootProject.ext.libs.rxjava_replayingshare
}
```

P.S. Would you share what have you changed in your fork?

MaxWebfactor, 2023-09-26:
Thanks for the advice, I will give it a shot!

_

> P.S. Would you share what have you changed in your fork?

_

It's a simple change to the service discovery function of the ServiceDiscoveryManager:

```
Single<RxBleDeviceServices> getDiscoverServicesSingle(final long timeout, final TimeUnit timeoutTimeUnit) {
        return deviceServicesObservable.doOnSubscribe(
                new Consumer<Disposable>() {
                    @Override
                    public void accept(Disposable disposable) {
                        timeoutBehaviorSubject.onNext(new TimeoutConfiguration(timeout, timeoutTimeUnit, Schedulers.computation()));
                    }
                });
    }
```

My goal is to remove the cached services to see if the operating system is caching the services of our Bluetooth device or not. I am currently developing an application where the smartphone is the Bluetooth peripheral device that communicates with the Bluetooth central device.
The central device is highly customizable, so its services can change at any time. 
On iOS this works fine (ok, I'm a Flutter developer, so I use the flutter_reactive_ble plugin from Philips Hue https://github.com/PhilipsHue/flutter_reactive_ble, which depends on your project), but on the Android side I've found that there is no way to reliably update the services. This means that I always get the same services, except that I have to reconnect the smartphone (programmatically reconnect works fine). This works, but has some unpleasant side effects. 

While researching where the cached services come from, I came across the service discovery feature mentioned above.

I also tried implementing this workaround that you discussed in 2018 - to no avail:
[Feature DiscoverService with clearCache #453 ](https://github.com/dariuszseweryn/RxAndroidBle/pull/453)