dariuszseweryn/RxAndroidBle #839: Set Preferred PHY / Read PHY

JamesDougherty, 2023-10-13:
We use this library for several of our projects and it's been great. However, our latest project is going to need the ability to set the PHY of the Bluetooth connection. It's currently using Nordic's Android Library, but we would like to fully switch over to this library for consolidation reasons. I've see this feature requested several times and I wasn't sure when it was going to be implemented, so I went ahead and forked the repo and added the functionality. Would it be possible to get this added into your repo so others can use the functionality and so I don't always have to worry merging in changes? If there is anything on my end that you need, then please let me know.

**Forked Repo**
[https://github.com/ChandlerSystems/RxAndroidBle/commit/3fc603873ddfa482644f4ea70eeefa09a25d0f1d](https://github.com/ChandlerSystems/RxAndroidBle/commit/3fc603873ddfa482644f4ea70eeefa09a25d0f1d)

**Example Use**
Here is an example of how it can be used. 

```java
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
    final int txPhy = RxBleConnection.PHY_LE_2M | RxBleConnection.PHY_LE_1M;
    final int rxPhy = RxBleConnection.PHY_LE_2M | RxBleConnection.PHY_LE_1M;
    final int phyOptions = RxBleConnection.PHY_OPTION_NO_PREFERRED;

    addDisposable(mConnectionObservable
            .flatMapSingle(connection -> connection.setPreferredPhy(txPhy, rxPhy, phyOptions))
            .take(1)
            .subscribe(
                (Boolean succeeded) -> Log.d(TAG,"PHY Set [Succeeded: " + succeeded + "]"),
                (Throwable throwable) -> Log.d(TAG,"On setPreferredPhy Error: " + throwable.getMessage())
            )
    );

    addDisposable(mConnectionObservable
            .flatMapSingle(RxBleConnection::readPhy)
            .take(1)
            .subscribe(
                (Pair<Integer, Integer> phy) -> Log.d(TAG, "PHY Read - Tx: " + phy.first + "  Rx: " + phy.second),
                (Throwable throwable) -> Log.d(TAG, "On setPreferredPhy Error: " + throwable.getMessage())
            )
    );
}
```

**Logs**
Here are some logs that show everything is working as expected.

_Android App Side_
[image]

_Firmware Side_
```txt
[00:26:33.326,721][DBG][BLE] 
[00:26:33.326,812][DBG][BLE] Device Connected - Connection Information
[00:26:33.326,904][DBG][BLE]   -- MAC Address: 65:42:4F:03:3E:DF
[00:26:33.326,995][DBG][BLE]   -- Type: LE Connection
[00:26:33.327,087][DBG][BLE]   -- Role: Peripheral
[00:26:33.327,178][DBG][BLE]   -- Tx Power: 20 dBm
[00:26:33.327,270][DBG][BLE]   -- Connection Interval: 36 ms
[00:26:33.327,392][DBG][BLE]   -- Peripheral Latency: 0 ms
[00:26:33.327,484][DBG][BLE]   -- Supervision Timeout: 5000 ms
[00:26:33.327,606][DBG][BLE] Watchdog Channel 2 Unregistered
[00:26:33.966,827][DBG][BLE] 
[00:26:33.966,949][DBG][BLE] Connection Parameters Updated
[00:26:33.967,041][DBG][BLE]   -- Connection Interval: 6 ms
[00:26:33.967,132][DBG][BLE]   -- Peripheral Latency: 0 ms
[00:26:33.967,224][DBG][BLE]   -- Supervision Timeout: 5000 ms
[00:26:34.258,850][DBG][BLE] 
[00:26:34.258,972][DBG][BLE] Connection Parameters Updated
[00:26:34.259,063][DBG][BLE]   -- Connection Interval: 36 ms
[00:26:34.259,155][DBG][BLE]   -- Peripheral Latency: 0 ms
[00:26:34.259,246][DBG][BLE]   -- Supervision Timeout: 5000 ms
[00:26:34.776,062][DBG][BLE] 
[00:26:34.776,153][DBG][BLE] LE PHY Updated
[00:26:34.776,245][DBG][BLE]   -- TX PHY: LE 2 Mbps
[00:26:34.776,336][DBG][BLE]   -- RX PHY: LE 2 Mbps
[00:26:34.866,394][DBG][BLE] 
[00:26:34.866,455][DBG][BLE] CCC Notification Status Change
[00:26:34.866,577][DBG][BLE]   -- Status: Enabled
[00:26:34.866,882][DBG][BLE] 
[00:26:34.866,943][DBG][BLE] LE Data Length (MTU) Updated
[00:26:34.867,218][DBG][BLE]   -- TX (Length: 251 bytes - Max Time: 2.12 ms (2120 us))
[00:26:34.867,431][DBG][BLE]   -- RX (Length: 251 bytes - Max Time: 2.12 ms (2120 us))
[00:26:34.920,654][DBG][BLE] 
[00:26:34.920,745][DBG][BLE] <<< TX  [Length: 94]
```

dariuszseweryn, 2023-10-15:
Hey, this looks very solid although there are a few nits I could make. Of course it could be merged — please create a Pull Request and we will discuss there.

JamesDougherty, 2023-10-16:
Hey Dariusz, appreciate it! I created the pull request, so I'll go ahead and close this and we can discuss any changes over there.

Pull Request: https://github.com/dariuszseweryn/RxAndroidBle/pull/840

closed 2023-10-16 by JamesDougherty