dariuszseweryn/RxAndroidBle #841: RxBleClient mulit connections

morkovkin, 2023-10-16:
Hello, thanks a lot for your library! I have issue in my project. I write program with support voice over ble and update remote control. So i have flow like this:
- When alc change i try find updates (for update i must take version from characteristic for compare with server bin file) and if i found it i start update, if not found just skip.
- I try to find specific characteristic in device and start voice service  in parallel with ota.

So, for example , if user use voice over ble,  how can i take version characteristic with out reconnect? 
if we can try to take version i think it's ok to continue to listen voice. But i don't understand how i can take 00002a28-0000-1000-8000-00805f9b34fb and continue listen voice from another service.

```kotlin
   class VoiceService : Service() {
       //singleton as write in docs
        @Inject
        lateinit var client: RxBleClient
        fun attachVoice() {
            //here we will change mtu take characteristic subscribe to characteristic chane
            client.getBleDevice(deivceMac).establishConnection().flatmap {
              connection.setupNotification(VoiceBytes)
            }
        }.subscribe({decode/transfer etc})
    }

    class UpdateService : Service() {
       //singleton as write in docs
        @Inject
        lateinit var client: RxBleClient

        fun checkUpdate() {
            //here we will change mtu take characteristic subscribe to characteristic chane
            client.getBleDevice(deivceMac).establishConnection().flatmap {
              connection.setupNotification(otaBytes).flatmap { device.writeCharacteristic(otaBytes)}
            }
        }

    }

    class AlcBroadcastReceiver : BroadcastReceiver() {
        override fun onReceive(context: Context?, intent: Intent?) {
            //some logic to take device
            
            //for examle we find device or lost device
            context.startService(VoiceService(device.mac))
            context.startService(UpdateService(device.mac))
        }
    }
```

dariuszseweryn, 2023-10-16:
Hello, thank you for using the library.
If I understood you correctly you find it difficult to use two concurrent services of your peripheral. If that's the case [this wiki](https://github.com/dariuszseweryn/RxAndroidBle/wiki/Tutorial:-Connection-Observable-sharing) page should help you.

morkovkin, 2023-10-17:
Hello! Thanks for the quick response! oh ya i forget about share, in this use case i can create some cache like
HashMap<Rx Device,RxConnection> and provide to my code..
Is it correct if i will write like this?

```kotlin
val cache = HashMap<RxBleDevice, Observable<RxBleConnection>>()

fun getConnection(device: RxBleDevice): Observable<RxBleConnection> {
    if (cache.contains(device)) return cache.get(device)
    val connection = device.establishConnection(false).compose(ReplayingShare.instance())
        .doOnError {
            //clean cache/etc
            cache.remove(device)
        }
    cache.put(device, connection)
    return connection
}

fun startVoiceService(device: RxBleDevice) {
    val disposable1 = getConnection(device).flatMap { connection ->
        //notification
        connection.setupNotification(UUID.randomUUID()).flatMap { observables ->
            //request like get voice codec etc
            connection.writeCharacteristic(
                UUID.randomUUID(),
                byteArrayOf()
            ).flatMapObservable {
                Observable.just(observables)
            }
        }.flatMap { it }
            .flatMap {
                //some logic
                if (it[0] == 0x0.toByte()) {
                    Observable.just(1)
                } else Observable.just(2)

            }

    }
}

fun startUpdateCheck(device: RxBleDevice) {
    val disposable2 = getConnection(device).flatMapSingle { connection ->
        //get sw version ect
        connection.readCharacteristic(UUID.fromString("00002a28-0000-1000-8000-00805f9b34fb"))
            .flatMap {
                if (it.littleEndianConversion() > 80) {
                    //logic to find update
                    Single.just(1)
                } else {
                    Single.just(2)
                }
            }
    }
}```

So, if disposable 2 is finished (because we was take info and do logic to find update its faster then wait user input as hid event) disposable 1 will still work?

dariuszseweryn, 2023-10-17:
Yes. It should still work

morkovkin, 2023-10-17:
Thank you for cool library!

closed 2023-10-17 by morkovkin