dariuszseweryn/RxAndroidBle #842: Is the library still maintained?

arbonvata, 2023-10-18:
The last release is almost one year old. Can't see much activity in the repo

dariuszseweryn, 2023-10-18:
Hello! 
While I do not have much time for maintenance nor I actively use the library — I try to address issues as they come and merge PRs. Do you have any specific needs?

arbonvata, 2023-10-20:
Actually it is more to see if we can use it our projects. Many ble libraries becomes obsolete with time due to lack of updates and breaking changes from google :)

dariuszseweryn, 2023-12-09:
I _hope_ this library will be obsolete because google would step-up and create an extension with a nice API support for older OS versions. Recently a [bluetooth Jetpack library](https://developer.android.com/jetpack/androidx/releases/bluetooth) was released, so 🤞.

closed 2023-12-09 by dariuszseweryn