dariuszseweryn/RxAndroidBle #843: Android 14 breaking change

angelix, 2023-11-06:
Android 14 introduced a [breaking change](https://developer.android.com/about/versions/14/behavior-changes-all#mtu-set-to-517) related to MTU negotiation.

This breaking change it may even require a firmware upgrade on the device to properly support the BLE spec in a more strict way.

On the library side, if you try to write more than 512 bytes using the `LongWriteOperationBuilder`, the write fails silently.

By default the following method is used to calculate the batch size.
```
//  MtuBasedPayloadSizeLimit.java
...
@Override
    public int getPayloadSizeLimit() {
        return rxBleConnection.getMtu() - gattWriteMtuOverhead;
    }
....
```

On Android 14 the call to `rxBleConnection.getMtu()` returns 517.
517 - 3 (GATT_WRITE_MTU_OVERHEAD) = 514 which is larger than the accepted 512.

Based on this [very informative issue ticket](https://issuetracker.google.com/issues/307234027) MTU Overhead should be 5.

As a work around you can use `setMaxBatchSize(512)` when creating the `LongWriteOperationBuilder` to use the `ConstantPayloadSizeLimit` implementation instead of the default `MtuBasedPayloadSizeLimit`.

angelix, 2023-11-12:
@e-Joe 

Example from [green_android](https://github.com/Blockstream/green_android/blob/fef44affbc8fd0b67ef64cf8c23d546a85897593/jade/src/main/java/com/blockstream/jade/JadeBleImpl.java#L174)
<pre>
@Override
    public int write(final byte[] bytes) {
        this.disposable.add(this.connection
                .flatMap(rxConn -> rxConn.createNewLongWriteBuilder()
                       <b> .setMaxBatchSize(512)</b>
                        .setCharacteristicUuid(IO_TX_CHAR_UUID)
                        .setBytes(bytes)
                        .build())
                .subscribe(this::onBytesSent,
                           this::onSendFailure)
        );
        return bytes.length;
    }
</pre>

sravan-wellnesys, 2023-12-05:
The work-around doesn't work in Android 14, using this **.setMaxBatchSize(512)** shows no effect for me, any solution for this yet?

angelix, 2023-12-05:
@sravan-wellnesys keep in mind that this change is for writing to BLE device. Be sure that the BLE device also respect the spec.

sravan-wellnesys, 2023-12-06:
> @sravan-wellnesys keep in mind that this change is for writing to BLE device. Be sure that the BLE device also respect the spec.

Can you please explain what exactly needs to be done, as I check this []( https://issuetracker.google.com/issues/307234027), Android 14 is by default requesting MTU 517, So should we update our peripheral device to respond to MTU 517 or By doing what you said **.setMaxBatchSize(512)** and configure our peripheral device to respond to MTU 512  even if Android 14 requests MTU 517?

sravan-wellnesys, 2023-12-06:
> > Can you please explain what exactly needs to be done, as I check this https://issuetracker.google.com/issues/307234027, Android 14 is by default requesting MTU 517, So should we update our peripheral device to respond to MTU 517 or By doing what you said **.setMaxBatchSize(512)** and configure our peripheral device to respond to MTU 512 even if Android 14 requests MTU 517?
> 
> Your device should respond to MTU request with the value 517, but the actual payload you send must be 512 bytes at maximum.
> 
> If the device sends more that 512 bytes, from my experience, Android will silently drop the packet.
> 
> On my case i had to firmware update my BLE device to send max 512 bytes, and my Android app to send max 512 bytes.
> 
> Explanation: MTU size includes the header size. The header can be from 3-5 bytes, but you should not really care about that. The crucial part is not to send more that 512 byte.
> 
> From the issue:
> 
> > A deeper reason is that the Bluetooth Specification allows the maximum size of an ATT attribute to be 512 bytes and the largest command ATT_PREPARE_WRITE_REQ has 5 bytes of header. Hence 512 + 5 = 517.
> > **The packet data itself should never exceed 512 as per the Bluetooth spec.**

Thanks a lot for your explanation, Is there a way that I don't send a value in requestMTU?, Do Android 14 still request 517 even if I don't specify a size in request instead of the default 23 bytes.

angelix, 2023-12-07:
> Thanks a lot for your explanation, Is there a way that I don't send a value in requestMTU?, Do Android 14 still request 517 even if I don't specify a size in request instead of the default 23 bytes.

Just don't call requestMTU at all.

From the issue:
> requestMTU(23) has the same effect as not calling requestMTU() at all, in that case the default MTU of 23 bytes will be used.

sravan-wellnesys, 2023-12-11:
> > > > Thanks a lot for your explanation, Is there a way that I don't send a value in requestMTU?, Do Android 14 still request 517 even if I don't specify a size in request instead of the default 23 bytes.
> > > 
> > > 
> > > Just don't call requestMTU at all.
> > > From the issue:
> > > > requestMTU(23) has the same effect as not calling requestMTU() at all, in that case the default MTU of 23 bytes will be used.
> > 
> > 
> > Yep, I am not receiving any bytes, not using requestMTU at all like you said. No data transfer is happening. So firmware upgrade is the only solution at this point? Btw my peripheral device is using Bluetooth 4.2.
> 
> I think so. Try using a different android version and see the size of data the device is sending. If it's more than 512, you can be sure that's your issue.

I tried with Android 12,  When Mtu - 512, I get 509 bytes, when Mtu size - 517, I receive 514 bytes, I thought the max was 512. Any reason why this is happening? can you please help.

sravan-wellnesys, 2023-12-12:
> I think is device issue that needs a fix with firmware update, the device should not send more than 512 bytes.

So anything more than 512 does not work for Android 14, but works for the previous versions?

angelix, 2023-12-12:
> So anything more than 512 does not work for Android 14, but works for the previous versions?

Correct, Android 14 is more strict following more strictly the version 5.2 of the Bluetooth Core.

dariuszseweryn, 2024-01-07:
This is a problem that the library can fix on its side — that is fix the LongWriteOperation (which btw. does _not_ use Prepared Write but chunks data on the application side) to send maximum of 512 bytes of data on Android 13 and above. That way people will not get caught by surprise when using LongWriteOperationBuilder's default chunk size. WDYT?

PR #847 (2024-02-03): Fixed CharacteristicLongWriteOperation defaults
On Android 13 when negotiated MTU > 515, CharacteristicLongWriteOperation would by default split data into chunks that did not fit Android internal data buffer for BLE writes. Reason for that is that on Android 13 the maximum attribute data buffer length was reduced from 600 to 512 bytes - which is now according to BLE Specs. Android 14 makes MTU negotiation to the maximum supported value on Android (517) by default, making this problem more likely to surface.

Fixes #843
closed 2024-02-03 by dariuszseweryn