dariuszseweryn/RxAndroidBle #844: Unable to automatically reconnect after 2 days or phone reboot

neowhoru, 2023-11-21:
**Describe the bug**
When your BLE devices is connected it works fine until the phone reboots or you wait 2 days without connecting.
Then you cannot reconnect automatically (you need to manually reconnect them/ disconnect).

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a BLE Device and stay connected with your app
2. Reboot your phone 
3. Your app is unable to autoconnect to the device anymore and gets the GATT 133 Error and a disconnectedException

**Expected behavior**
The connection state of the ble device should be connected.

**Smartphone (please complete the following information):**
 - Device: Pixel 5 
 - OS: Android 14 (but also doesnt work on Samsung Android 13 Devices)
 - Library version: 1.17.2

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
To turn on logs use:
```
 MAC='##:##:##:##:##:##'  onConnectionStateChange(), status=133, value=0
 
 Connection operations queue to be terminated (MAC='D5:A7:1F:A7:AF:BD')
 
com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='##:##:##:##:##:##' with status 133 (GATT_ERROR)
	at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
	at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:340)
	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:936)
	at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback(Unknown Source:0)
	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:335)
	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:211)
	at android.os.Binder.execTransactInternal(Binder.java:1344)
	at android.os.Binder.execTransact(Binder.java:1275)
```

dariuszseweryn, 2024-02-03:
Closing due to lack of information and inactivity of the original poster.

When more data will be available—feel free to add that information and reopen this issue.

Best Regards

closed 2024-02-03 by dariuszseweryn

neowhoru, 2024-02-05:
Hello Darius,
sorry i just was out of order a bit. 
I will try to answer:

> Hello, two questions:
> 
>     * is your peripheral bonded?
> 
>     * does your app scan the peripheral before trying to connect?
> 
> 
> I guess that to both questions the answer is: no. There is a (pretty outdated, but still valid) [wiki section about this problem](https://github.com/dariuszseweryn/RxAndroidBle/wiki/FAQ:-Cannot-connect#connect-without-a-scan).

Yep it was bounded before and normally it does a scan but not on devices that are "already bounded". 
However we found out its not an issue with RxAndroidBle - its an issue with the Android Ble Stack and the Google Pixel 5.

Also you don't need to wait 2 days.

To reproduce the issue:
- bound a device with your custom app as usual 
- on your google pixel 5 disable bluetooth and re-enable it
- your bounded devices is no longer in the "bounded" device list and not bounded anymore

We don't know why and we don't know which devices this occurs but on Samungs devices for example same app same device works pretty well.

I just post here for the guys who might search for the same :)

dariuszseweryn, 2024-02-05:
Good that you've pinpointed the issue. You may probably workaround it by always scanning your peripheral before attempting to connect.