dariuszseweryn/RxAndroidBle #846: Unable to readCharacteristic after writeCharacteristic

Unai-HP, 2024-01-07:
Hi! Firstly, thank you for this amazing library. I've been trying to fix some BLE connectivity issues in the Wear OS app of the OSS project called [xDrip](https://github.com/NightscoutFoundation/xDrip)  since last month. xDrip uses rxandroidble (v2) and I think I've been able to identify where it's stucked. There are no problems connecting to the device but the in code below:

```
// With this method we try to authenticate the ble device
// Set AuthRequestByteSequence
connection.writeCharacteristic(Authentication, AuthRequestByteSequence)
        .subscribe(
                characteristicValue -> {
                        // characteristicValue is always recieved

                        connection
                                .readCharacteristic(Authentication)
                                .subscribe(
                                        readValue   -> { /** Never reads the value */ }, 
                                        throwable   -> { /** It always throws with GATT_SUCCESS status 0 */
                                });
                },
                throwable -> { /** Never happends */ }
        );
```

After the second subscribe, it should be able to receive "readValue". However, it always executes the "throwable" part, with GATT_SUCCESS. I'm an absolute noob to BLE and I'm blocked (I haven't found anything regarding this in Google/StackOverflow), so I've thought of opening this and maybe get some help. Sorry if this is not how/where is asked, I'll remove the issue as soon as possible. Sorry for my bad English, It's not my first language.

The throwable message its this:
```
com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS) 
```

dariuszseweryn, 2024-01-08:
Hello, thanks for using the library!

Unfortunately you didn't provide much context in the logs so I can only guess that the connection is not being kept alive. Connection should be kept alive as long as the subscribers to `RxBleDevice#establishConnection` are subscribed.

Is it the case?

Unai-HP, 2024-01-09:
Hi. Thanks for replying so fast!
You are right, I didn't provide enough info, so I've rechecked to paste the logs here and found some (more descriptive) errors. I don't know how I missed them.
```SHELL
D  setCharacteristicNotification() - uuid: f8083535-849e-531c-c594-30f1f86a4ea5 enable: true
D  QUEUED   CharacteristicWriteOperation(125660900)
D  STARTED  CharacteristicWriteOperation(125660900)
I  RUNNING  CharacteristicWriteOperation{MAC='XX:XX:XX:XX:XX:XX', characteristic=[uuid='...', hexValue=[...]]}
I  MAC='XX:XX:XX:XX:XX:XX'    onCharacteristicWrite(), status=0, value=[uuid='...']

W  Characteristic ... supports properties: [ WRITE INDICATE ] (40) does not have any property matching [ READ ] (2) <--- The error

D  QUEUED   CharacteristicReadOperation(91684627)
D  FINISHED CharacteristicWriteOperation(125660900) in 442 ms
D  STARTED  CharacteristicReadOperation(91684627)
I  RUNNING  CharacteristicReadOperation{MAC='XX:XX:XX:XX:XX:XX', characteristic=[uuid='...']}

D  Could not read after AuthRequestTX: com.polidea.rxandroidble2.exceptions.BleGattCannotStartException: 
GATT exception from MAC address 'XX:XX:XX:XX:XX:XX', with type BleGattOperation{description='CHARACTERISTIC_READ'}

D  FINISHED CharacteristicReadOperation(91684627) in 37 ms
I  MAC='XX:XX:XX:XX:XX:XX'  onConnectionStateChange(), status=0, value=0
D  Connection operations queue to be terminated (MAC='XX:XX:XX:XX:XX:XX')
   com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS)
   	at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
   	at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:307)
   	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:874)
   	at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback(Unknown Source:0)
   	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:302)
   	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:128)
   	at android.os.Binder.execTransactInternal(Binder.java:1285)
   	at android.os.Binder.execTransact(Binder.java:1244)
D  Connection Disconnected/Failed: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS)
D  QUEUED   DisconnectOperation(4749679)
D  setCharacteristicNotification() - uuid: f8083535-849e-531c-c594-30f1f86a4ea5 enable: false
D  STARTED  DisconnectOperation(4749679)
I  RUNNING  DisconnectOperation{MAC='XX:XX:XX:XX:XX:XX'}
D  onConnectionStateChange newState: RxBleConnectionState{DISCONNECTING}
```
Looks like the "read" operation doesn't exist, I guess I'm not doing the auth correctly. I've seen some post of this already, so I'll check those to see if I can get a fix. Again, thanks for your get.

dariuszseweryn, 2024-02-03:
@Unai-HP Did you happen to fix the issue?

Unai-HP, 2024-02-03:
Yes, I forgot to close the thread, sorry.

Unai-HP, 2024-02-03:
Yes, I forgot closing the issue, sorry.

closed 2024-02-03 by Unai-HP

dariuszseweryn, 2024-02-03:
I am glad that you were able to find a fix. 

If you can disclose it — please post the root cause and solution for the problem — other people who would come here later could benefit from that. Thanks in advance!