dariuszseweryn/RxAndroidBle #851: In the case of unsuccessful Bluetooth connection, actively disconnect without executing context. unregisterReceiver (receiver); At this point, there may be multiple Bluetooth broadcast registrarReceivers, which may crash on some Huawei phones due to a broadcast limit of approximately 1000. Suggest fixing this issue.

stylsw, 2024-03-20:
After calling establishConnection, it was found that the context. registerReceiver (receiver, new IntentFilter (BluetoothAdapter. ACTION_STATE.CHANGED) in RxBleAdapterState Observable) would be triggered;

In the case of a successful Bluetooth connection, actively disconnecting through dispose will execute context. unregisterReceiver (receiver);

If the Bluetooth connection is not successful at this time, actively disconnect through dispose without executing context. unregisterReceiver (receiver); At this point, there may be multiple Bluetooth broadcast registrarReceivers, which may crash on some Huawei phones due to a broadcast limit of approximately 1000. Suggest fixing this issue

dariuszseweryn, 2024-03-20:
Hello,
Thanks for using the library and for the report. Could you provide a step by step reproduction scenario and some logs that support the case so I could act? A memory dump showing leaks could also be useful. Thanks!

stylsw, 2024-03-22:
Hello, you can use a fake device to execute the establishConnection method, then perform a dispose operation before the connection is unsuccessful, and log before the unregisterReceiver of RxBleAdapterState Observable. You will find that the unregisterReceiver execution has not been triggered. This is a must see scenario.

ArturMytsko, 2025-02-24:
Hi @dariuszseweryn 
I have the same issue on my project. App is trying to connect to device in backgound with timeout, and it reaches the limit of 1000 registered broadcast receivers after some time. This is the general limit in Android, and could be reproduced **on any device!**.

As pointed by @stylsw, the problem is happening when calling **establishConnection** and dispose when device is not connected. It seems that the problem hides inside "RxBleAdapterStateObservable", when the library registers receiver but unregister never inovked.

Here is the simple code exmaple you can use to reproduce the issue:
```
import android.content.Context
import com.polidea.rxandroidble3.RxBleClient
import io.reactivex.rxjava3.core.Completable
import io.reactivex.rxjava3.disposables.Disposable
import io.reactivex.rxjava3.schedulers.Schedulers
import java.util.concurrent.TimeUnit
import timber.log.Timber

class RxBleLibraryTest() {
    private var connectionDisposable: Disposable? = null
    private var connectionsCount: Int = 0
    private var rxBleClient: RxBleClient? = null
    private var timeout: Long = 50L // ms
    private var retryDelay: Long = 50L // ms

    fun startTest(context: Context, timeout: Long = 50L, retryDelay: Long = 50L) {
        rxBleClient = RxBleClient.create(context)
        this.timeout = timeout
        this.retryDelay = retryDelay
        Timber.i("Start Connections test, timeout = $timeout, retryDelay = $retryDelay")
        startScanning()
    }

    fun startScanning() {
        rxBleClient?.let { bleClient ->
            val bleDevice = bleClient.getBleDevice("11:22:33:44:55:55")
            connectionDisposable = bleDevice.establishConnection(false)
                .timeout(timeout, TimeUnit.MILLISECONDS)
                .subscribeOn(Schedulers.io())
                .observeOn(Schedulers.io())
                .subscribe({
                }, { throwable: Throwable? ->
                    Timber.i("Connections count = $connectionsCount")
                    connectionsCount++

                    Timber.e(throwable)

                    connectionDisposable?.dispose()
                    Completable.timer(retryDelay, TimeUnit.MILLISECONDS).blockingAwait()
                    startScanning()
                })
        }
    }
}
```

And start the test in your activity:
```
 RxBleLibraryTest().startTest(this)
```

After reaching 1000 attempts, the app start logging this error message:
```
Failed to monitor adapter state.
java.lang.IllegalStateException: Too many receivers, total of 1000, registered for pid: 6775, callerPackage: com.test.app
	at android.os.Parcel.createExceptionOrNull(Parcel.java:3019)
	at android.os.Parcel.createException(Parcel.java:2995)
	at android.os.Parcel.readException(Parcel.java:2978)
	at android.os.Parcel.readException(Parcel.java:2920)
	at android.app.IActivityManager$Stub$Proxy.registerReceiverWithFeature(IActivityManager.java:5017)
	at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1816)
	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1750)
	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1738)
[... 41 lines cut ...]
	at bleshadow.dagger.internal.DoubleCheck.get(DoubleCheck.java:47)
	at com.polidea.rxandroidble3.DaggerClientComponent$ConnectionComponentImpl.connectionSubscriptionWatchers(DaggerClientComponent.java:367)
	at com.polidea.rxandroidble3.internal.connection.ConnectorImpl.lambda$prepareConnection$2$com-polidea-rxandroidble3-internal-connection-ConnectorImpl(ConnectorImpl.java:42)
	at com.polidea.rxandroidble3.internal.connection.ConnectorImpl$$ExternalSyntheticLambda0.get(D8$$SyntheticClass:0)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDefer.subscribeActual(ObservableDefer.java:33)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13263)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoFinally.subscribeActual(ObservableDoFinally.java:43)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13263)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDefer.subscribeActual(ObservableDefer.java:40)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13263)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableTimeoutTimed.subscribeActual(ObservableTimeoutTimed.java:47)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13263)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
	at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:80)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:71)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:307)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
	at java.lang.Thread.run(Thread.java:1012)
```

Please take a look.
If you have a any idea of the fix without updating the library - let us know.

ArturMytsko, 2025-05-19:
@dariuszseweryn  any updates?

dariuszseweryn, 2025-05-30:
I managed to sit for a moment yesterday on the problem — IT seems that DisconnectionRouter is not unsubscribing from the state observable. Still need to fix it

dariuszseweryn, 2025-06-04:
Aaaand is gone. I'll probably make a new release in the next few days

PR #864 (2025-06-04): Fixed RxBleAdapterStateObservable leak.
Cause: the DisconnectionRouter did not properly cleaned up its state as long as neither the BluetoothGattCallback signalled a disconnection or Android did broadcast change in the BluetoothAdapter state. For cases when the connection unsubscription happened before the BluetoothGatt did started to connect (before BluetoothGattCallback#onConnectionStateChange was called, <~50 milliseconds) neither of these cases had place, causing a leak.

Fixes #851
closed 2025-06-04 by dariuszseweryn