dariuszseweryn/RxAndroidBle #854: Add `advertisingSid` property to `ScanResult` and `RxBleInternalScanResult`

KrzysztofMamak, 2024-05-23:
**Is your feature request related to a problem? Please describe.**
`com.polidea.rxandroidble2.scan.ScanResult` does not contain an information about advertising sid which is needed for supporting Bluetooth Auracast feature (https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/24670-BASS-html5/out/en/index-en.html#UUID-2a17a3a2-36f9-8cb4-0877-3b56055a7955). It means that there is no option to retrieve this information from the `rxandroidble` library.

**Describe the solution you'd like**
`com.polidea.rxandroidble2.scan.ScanResult` and `com.polidea.rxandroidble2.internal.scan.RxBleInternalScanResult` classes can be extended with `advertisingSid` property. This value can be retrieved using `android.bluetooth.le.ScanResult.getAdvertisingSid` method (https://developer.android.com/reference/kotlin/android/bluetooth/le/ScanResult#getadvertisingsid) and propagated in `com.polidea.rxandroidble2.internal.scan.InternalScanResultCreator`.

I can volunteer and submit a PR soon.

dariuszseweryn, 2024-06-22:
Hey, sorry for a long pause, got a busy time at work. Feel free to open a PR 👍

KrzysztofMamak, 2024-07-11:
I'll open a PR soon.

dariuszseweryn, 2024-07-12:
I have one request, when adding tests please use JUnit – I've started migrating to it from Spock as it's not supported for Android anymore.

PR #856 (2024-09-09): Add `advertisingSid` property to `ScanResult`
Add `advertisingSid` property to `ScanResult`.

Fixes #854
closed 2024-09-09 by dariuszseweryn