dariuszseweryn/RxAndroidBle #859: UndeliverableException thrown  / OnCompletedFailedException - Help needed

cogesmobilesolutions, 2024-09-05:
**Describe the bug**
We been receiving so many crash for UndeliverableException.

We are recreating the client everytime when connect the device using the code snippet below.

`rxBleClient = RxBleClient.create(this.context)`

But the documentation suggest, **It's your job to maintain single instance of the client** so, by moving the client initialation 
to the application class will help to reduce the problem?

Also i attach a stack trace as well.

`          Caused by rx.exceptions.OnCompletedFailedException: Task lk0$a@d22a638 rejected from java.util.concurrent.ThreadPoolExecutor@8ef16aa[Shutting down, pool size = 1, active threads = 1, queued tasks = 0, completed tasks = 2]
       at rx.observers.SafeSubscriber.onCompleted(SafeSubscriber.java:85)
       at rx.internal.operators.OperatorUnsubscribeOn$1.onCompleted(OperatorUnsubscribeOn.java:42)
       at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.onCompleted(OperatorSubscribeOn.java:89)
       at rx.internal.operators.OnSubscribeCreate$BaseEmitter.onCompleted(OnSubscribeCreate.java:97)
       at com.polidea.rxandroidble.internal.operations.DisconnectOperation.considerGattDisconnected(DisconnectOperation.java:106)
       at com.polidea.rxandroidble.internal.operations.DisconnectOperation$1.onError(DisconnectOperation.java:87)
       at rx.internal.util.ObserverSubscriber.onError(ObserverSubscriber.java:39)
       at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:153)
       at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
       at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:60)
       at rx.internal.operators.OnSubscribeLift.call(OnSubscribeLift.java:30)
       at rx.Observable.subscribe(Observable.java:10352)
       at rx.Observable.subscribe(Observable.java)
       at rx.Observable.subscribe(Observable.java:10227)
       at com.polidea.rxandroidble.internal.operations.DisconnectOperation.protectedRun(DisconnectOperation.java:75)
       at com.polidea.rxandroidble.internal.QueueOperation$1.call(QueueOperation.java:42)
       at com.polidea.rxandroidble.internal.QueueOperation$1.call(QueueOperation.java:38)
       at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:72)
       at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:32)
       at rx.Observable.unsafeSubscribe(Observable.java:10256)
       at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.call(OperatorSubscribeOn.java:100)
       at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
       at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
       at java.lang.Thread.run(Thread.java:1012)
        `
Regards.

dariuszseweryn, 2024-09-06:
TBH I have not seen this problem yet. Keeping a single client could help as all required thread pools won't get deallocated.

cogesmobilesolutions, 2024-09-16:
@dariuszseweryn Thank you for your reply, we will keep single client and let you know the feedback.