dariuszseweryn/RxAndroidBle #865: ScanFilter for FEAA UUID not working in RxAndroidBle 1.16.0 (extended advertising)

Megha-Shah-Einfochips, 2025-09-18:
- I am using com.polidea.rxandroidble2:rxandroidble:1.16.0.
- I’m trying to scan for Google Finder / Nearby devices that advertise the FEAA service UUID (0000FEAA-0000-1000-8000-00805F9B34FB) using RxAndroidBle 1.16.0 on Android 10–14.
- Native BluetoothLeScanner successfully finds FEAA devices with the correct ScanFilter and setLegacy(false).
- Using RxAndroidBle, the scan finds the device, but the service data for FEAA is always null if a ScanFilter with the
UUID is applied.
- Removing the UUID filter and manually checking scanResult.getScanRecord().getServiceData(feaaUuid) works, but this defeats the purpose of filtering in the library.
- setLegacy(false) does not seem to have an effect when using ScanFilter in RxAndroidBle.

- Steps to Reproduce:
   ParcelUuid feaaUuid = ParcelUuid.fromString("0000FEAA-0000-1000-8000-00805F9B34FB");
   ScanFilter filter = new ScanFilter.Builder()
        .setServiceUuid(feaaUuid)
        .build();
   ScanSettings settings = new ScanSettings.Builder()
        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
        .setLegacy(false)
        .build();
    rxBleClient.scanBleDevices(Collections.singletonList(filter), settings)
    .subscribe(scanResult -> {
        // scanResult.getScanRecord().getServiceData(feaaUuid) is always null
    });

- Expected behavior:
  The scan should return FEAA devices and provide their service data.

- Actual behavior:
   Scan either returns devices without service data, or does not find devices at all if the UUID filter is applied.

- Environment:
  RxAndroidBle: 1.16.0
  Android versions: 10, 11, 12, 13, 14
  Tested on multiple devices and emulators
  Permissions correctly granted (BLUETOOTH_SCAN, BLUETOOTH_CONNECT, ACCESS_FINE_LOCATION)

- Additional Notes:
Native BluetoothLeScanner works correctly for FEAA when setLegacy(false) and ScanFilter with the UUID is applied.
The issue seems specific to RxAndroidBle handling of extended advertising + UUID filters.

dariuszseweryn, 2025-09-19:
Hey, thanks for using the library. Have you examined what is returned in the `ScanRecord`?

Megha-Shah-Einfochips, 2025-09-19:
Yes, it return all nearby devices like FE2C, FE09, FEAB,FEAC

Megha-Shah-Einfochips, 2025-09-19:
```java
{
    "count": "25",
    "devices": [
        {
            "advertise": "02 01 1A 0D FF 4C 00 16 08 00 5F B7 AB 4C 96 25 DE ",
            "mac": "5A:B5:F1:70:64:68",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 02 18 16 F1 FC 04 71 D3 DC 62 C2 49 9C 1D 5A 2A A2 4B A4 86 81 18 85 F9 6C 82 ",
            "mac": "3A:E5:52:F0:79:6D",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 18 03 02 BE FE 04 16 8F FC 00 0F 09 42 6F 73 65 5F 42 69 6C 6C 69 65 5F 43 31 ",
            "mac": "E4:58:BC:2A:AB:38",
            "name": "_Billie_C1"
        },
        {
            "advertise": "02 01 1A 02 0A 0C 0C FF 4C 00 10 07 74 1F 3E 68 19 8A A8 ",
            "mac": "67:54:50:AC:02:E5",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 04 1A FF 00 4C 02 15 26 86 F3 9C BA DA 46 58 85 4A A6 2E 7E 5E 8B 8D 00 01 00 00 C9 ",
            "mac": "37:70:E4:43:3E:9F",
            "name": "< Product>"
        },
        {
            "advertise": "18 FF 06 00 34 F0 00 2A 59 21 E5 BD 0F 26 48 B1 A1 86 62 17 32 5C 87 70 3F ",
            "mac": "29:F3:F3:F0:C0:13",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 18 03 02 BE FE 13 09 42 6F 73 65 20 51 43 20 55 6C 74 72 61 20 32 20 48 50 ",
            "mac": "63:C5:91:58:3A:F9",
            "name": " QC Ultra 2 HP"
        },
        {
            "advertise": "02 01 18 10 16 2C FE 10 40 88 52 00 22 21 F1 FC 34 FF FF FF 02 0A F4 13 09 42 6F 73 65 20 51 43 20 55 6C 74 72 61 20 32 20 48 50 ",
            "mac": "5C:7F:3A:36:91:A5",
            "name": " QC Ultra 2 HP"
        },
        {
            "advertise": "02 01 1A 02 0A 0C 0A FF 4C 00 10 05 5D 1C B6 AE AF ",
            "mac": "48:4B:20:39:5C:97",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 06 19 16 F7 FD 01 3F 12 18 29 4E 4E 5B 67 69 B7 41 50 8C 94 7D 4D 00 00 00 00 03 03 03 78 FE 06 FF 65 00 01 C9 05 ",
            "mac": "3A:CA:84:5B:40:32",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 02 19 16 F1 FC 04 A9 68 22 55 01 69 7F FA F7 BD 93 33 BD F5 2E C4 A9 DD 2A A4 C5 ",
            "mac": "1D:24:2C:85:F0:15",
            "name": "< Product>"
        },
        {
            "advertise": "1E 16 F3 FE 4A 17 23 30 32 49 57 11 32 44 43 3C 16 A9 65 64 1E 8D 6E 6A E1 2C 3B 58 C3 CB D5 ",
            "mac": "15:F6:E8:BA:D9:B2",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 02 18 16 F1 FC 04 55 C3 2F 3C 0D 4E 03 68 C6 7C 80 87 BC 8C AE 35 22 9D CE BF ",
            "mac": "0F:1A:28:9D:C2:32",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 1A 02 0A 08 0B FF 4C 00 10 06 37 1A C0 1C A1 DA ",
            "mac": "71:77:F4:08:1F:5E",
            "name": "< Product>"
        },
        {
            "advertise": "07 FF 4C 00 12 02 00 03 ",
            "mac": "FD:11:4D:8B:26:9A",
            "name": "< Product>"
        },
        {
            "advertise": "07 FF 4C 00 12 02 00 03 ",
            "mac": "ED:46:F7:E1:D9:B1",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 06 05 16 61 FE 00 02 05 FF DA 01 00 02 13 09 4D 65 65 74 55 70 20 53 6F 66 74 20 52 65 6D 6F 74 65 ",
            "mac": "44:73:D6:D9:30:94",
            "name": "MeetUp Soft Remote"
        },
        {
            "advertise": "07 FF 4C 00 12 02 00 02 ",
            "mac": "D6:37:2F:32:A5:4B",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 06 16 FF 1B 01 08 3D 74 D5 65 DE 28 19 41 B9 83 01 00 00 66 5D 00 00 FF ",
            "mac": "28:DE:65:D5:74:3D",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 06 16 FF 1B 01 08 33 74 D5 65 DE 28 19 41 B9 83 01 00 00 98 5D 00 00 FF ",
            "mac": "28:DE:65:D5:74:33",
            "name": "< Product>"
        },
        {
            "advertise": "07 FF 4C 00 12 02 00 02 ",
            "mac": "D7:1D:12:D4:A7:69",
            "name": "< Product>"
        },
        {
            "advertise": "07 FF 4C 00 12 02 00 02 ",
            "mac": "F3:F6:63:6E:4A:BB",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 06 16 FF 1B 01 08 41 74 D5 65 DE 28 19 41 B9 83 01 00 00 70 5D 00 00 FF ",
            "mac": "28:DE:65:D5:74:41",
            "name": "< Product>"
        },
        {
            "advertise": "59 16 F3 FE 4A 17 23 30 32 49 57 11 32 44 43 3C 16 A9 65 64 1E 8D 6E 6A E1 2C 3B 58 C3 CB D5 02 39 A0 17 8D 84 F9 9F 39 4E D2 1D 78 3C 85 86 0D 5E 14 46 AF 47 94 CA 4E 24 EA B0 F3 C5 53 79 10 9D 33 48 84 15 05 7F 00 00 84 15 06 61 00 00 84 15 04 7F 00 00 83 15 01 FF FF ",
            "mac": "07:FD:D3:D0:56:21",
            "name": "< Product>"
        },
        {
            "advertise": "02 01 06 16 FF 1B 01 08 33 74 D5 65 DE 28 19 41 B9 83 01 00 00 9D 5D 00 00 FF ",
            "mac": "28:DE:65:D5:74:33",
            "name": "< Product>"
        }
    ]
}
```  
here I attached my scanResult.

Megha-Shah-Einfochips, 2025-09-22:
@dariuszseweryn Can you please give me a information who handle this issue?

Megha-Shah-Einfochips, 2025-09-30:
Hi @dariuszseweryn, After further analysis, it seems that the RXAndroidBle library scanner expects the FMDN frame(Google Finder) in a specific format. When I put byte 0x40 (tracking protection off) or 0x41 (tracking protection on) after the service data UUID (0xFEAA) in the advertisement, the scanner is able to detect the frame. So for now we will put this byte after the service data UUID in the advertisement. The main thing to check is if we really have this filter mechanism in the RXAndroidBle library, if yes, I believe it should not be. We need to pass the data as received to the application layer.